ClusterDuck-Protocol/ClusterDuck-Protocol #9: Update the README and PDF to specify exact ZIP file to import

krook, 2020-03-11:
**Describe the solution you'd like**
Update the README and Make a Duck Manual to mention the release ZIP file as the thing that can be imported into the Arduino environment (in addition to the source copying option). Right now it just says there's a ZIP file (which can be confused for Libraries.zip inside the repo).

This can point to either the stable release or latest from the master branch, depending on which the user wants to run:

- https://github.com/Code-and-Response/ClusterDuck-Protocol/releases
- https://github.com/Code-and-Response/ClusterDuck-Protocol/archive/master.zip

TimoWielink, 2020-03-11:
@krook Please review the Add zip option

krook, 2020-03-12:
Excellent, I see the [commit](https://github.com/Code-and-Response/ClusterDuck-Protocol/commit/f0e331457527cedf618b520a13841ddd830ac4cf)

closed 2020-03-19 by TimoWielink