ClusterDuck-Protocol/ClusterDuck-Protocol #10: add source for generating the pdf

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
Would like to help update the info in the .pdf, but do not know where the source for it is. Can that be shared? 

**Describe the solution you'd like**
The "source" for the pdf to be shared so it can be updated. (even if it is a word doc or something like that)

**Describe alternatives you've considered**
The pdf may be able to be generated from doxygen/sphinx or other toolings.

**Additional context**
Just want to help improve this cool project. :-)

krook, 2020-03-16:
@TimoWielink thoughts on this one? Maybe there's something from our Call for Code kit scripts that can be reused @derekteay? https://github.com/Call-for-Code/Digital-Kits/tree/master/scripts

TimoWielink, 2020-03-16:
let me take a look at this.

Salzi, 2020-03-28:
Is there a reason why this is a pdf? Wouldn't it be much easier/maintainable to do this with markdown - maybe even as a wiki in this repository here

knouse1344, 2020-03-28:
is the "pdf" referenced here the make a duck manual?

Salzi, 2020-03-28:
Yes it is linked in the [readme](https://github.com/Code-and-Response/ClusterDuck-Protocol/blob/master/README.md#api-and-getting-started) but its a link to a google document (and an older version)

mkinney, 2020-03-28:
The google docs is a fine option, thanks. 

I'm ok with closing this issue, unless anyone else wants to migrate that doc into this repo and use markdown or something.

TimoWielink, 2020-03-28:
We might add as a markdown, since the document is also used on the website we wanted to keep it easy to access from multiple sources

knouse1344, 2020-03-29:
agreed.  just make sure its all up to date @TimoWielink .   Markdown is good for github and particularly coders/nerds.... though i still like having a pdf for anyone who isn't very experienced with github to be able to grab, read, and forward along.

Salzi, 2020-03-29:
Just tried to create a markdown version of that How To [here](https://github.com/Salzi/ClusterDuck-Protocol/blob/markdownDoc/doc/howToBuildADuck.md)

From there we could use github actions to generate a pdf from that source. For example with [pandoc](https://github.com/pandoc/pandoc-action-example)

Salzi, 2020-03-29:
Did some experiments locally.

Generated the pdf with:

`pandoc howToBuildADuck.md --pdf-engine=xelatex  -V 'mainfont:DejaVu Sans' -V 'geometry: left=2cm,right=2cm,top=3cm,bottom=3cm' -o out.pdf`

And added the title page (manually). This is the result: 
[out.pdf](https://github.com/Code-and-Response/ClusterDuck-Protocol/files/4398890/out.pdf)

I'm pretty sure we can do this automatically.

Salzi, 2020-03-29:
Sorry for spam...

[Here](https://github.com/Salzi/ClusterDuck-Protocol/actions/runs/65865150) is the fully automated process to generate the PDF documentation out of the markdown file

PR #30 (2020-03-31): How to build a duck as markdown
Add the how to build a duck as markdown and add a github workflow to build the PDF.

Should fix  #10
closed 2020-03-31 by TimoWielink