ClusterDuck-Protocol/ClusterDuck-Protocol #11: add comments to all functions

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
Add comments to all functions. Bonus points if they are doxygen-compatible.

**Describe the solution you'd like**
Add/format comments on each function in ClusterDuck code.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Doxygen-based formatting is probably the best option, but might consider using something like https://www.sphinx-doc.org

**Additional context**
Start small. Add comments and some auto-doc gen stuff.

krook, 2020-04-09:
This is being addressed partially in this [work in progress](https://github.com/Code-and-Response/ClusterDuck-Protocol/pull/37) pull request.

nfeuer, 2020-11-30:
This has been added with the Doxygen update

closed 2020-11-30 by nfeuer