ClusterDuck-Protocol/ClusterDuck-Protocol #13: Add static code analysis tooling

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
Add some sort of lint/static code analysis

**Describe the solution you'd like**
Create a way to run some static code analysis on the code via command line.

**Describe alternatives you've considered**
Just a quick google shows these projects:
https://clang.llvm.org/extra/clang-tidy/
http://cppcheck.sourceforge.net/

**Additional context**
Would be nice to then connect this to either https://travis-ci.org/ or use a github action.

mkinney, 2020-03-14:
Might borrow pieces from https://github.com/me-no-dev/ESPAsyncWebServer

nfeuer, 2020-06-18:
@hawk4031 as we make changes this would be a good help. It would also help prevent issues that @TimoWielink had earlier with the `runSensor`.

closed 2021-06-07 by nfeuer