ClusterDuck-Protocol/ClusterDuck-Protocol #14: Improve the Arduino library import process

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
There are libraries in a zip file in the repo. This is not a "best practice", but completely understand why/how it is the way it is.

**Describe the solution you'd like**
Create project files/some way to refer to other projects.

**Describe alternatives you've considered**
Ensure there are minimal binary/zip files in repo. Document steps to create a simple Mama/Papa from nothing (in regards to the library files)

**Additional context**
Looks like you can create dependencies (see https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ )

krook, 2020-03-16:
Related to #9

Salzi, 2020-03-29:
@TimoWielink Did #29 and #32 solves this?

TimoWielink, 2020-03-29:
@Salzi yes!

krook, 2020-03-31:
Both PRs have landed. Closing.

closed 2020-03-31 by krook