ClusterDuck-Protocol/ClusterDuck-Protocol #17: Template for printing stickers for case?

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
Is there a doc/pdf for printing the stickers on the outside of the plastic cases?

**Describe the solution you'd like**
Share location to the file(s) used to print for outside of case.

**Describe alternatives you've considered**
Would be neat to add a 2d QR code that people could scan that would take them to the website.

**Additional context**
Might help with promotion/discussions/awareness.

TimoWielink, 2020-03-18:
I like this idea, I can definitely upload a sticker I made for one of our first Ducks. What kind of information would be useful to put on the sticker?

mkinney, 2020-03-18:
What you used would be fine. Please share the pdf and/or source docs.

If a local situation happened (or if I wanted to deploy some ducks), I would want something on the case that would identify it as "not something suspect". (if that makes sense)

A logo, link to website and possibly a 2d QR code or something like that.

TimoWielink, 2020-03-26:
There is a Sticker file in the asset folder.

mkinney, 2020-03-28:
Great. Thanks!

What software was used to create that jpg? Could that file also be added? (just thinking if someone wanted to slightly alter it)

But, if not, no biggie.

TimoWielink, 2020-03-28:
I have a psd File can share as well.

mkinney, 2020-03-29:
That'd be great! Thanks.

krook, 2020-03-31:
@TimoWielink is there an SVG version?

krook, 2020-04-09:
@TimoWielink do you have a vector format or SVG?

TimoWielink, 2020-04-09:
Yes I can provide that

PR #38 (2020-04-09): Ad svg for sticker 
fix #17 add svg for sticker for the duck cases.
closed 2020-04-09 by TimoWielink