ClusterDuck-Protocol/ClusterDuck-Protocol #18: add info about distances/speed (actual vs. theoretical)

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
Can info be shared about what was learned during some deployments? (ex: 400m with clear line of sight between MomaDucks got this much bandwidth/error rate, etc.)

**Describe the solution you'd like**
Some info on what was found during deployments. Perhaps what is "optimal/maximal" vs. actual.

**Describe alternatives you've considered**
WiFi should be fast (somewhat normal 2.4 wifi network speeds) between DuckLinks, but from Mama to Papa, since going over 900MHz is limited. Document/share these speeds/info.

**Additional context**
Might be cool to show info on how using pringle cans and/or low-tech antennas might improve. (like lofting some Mama's via balloons in the video)

krook, 2020-04-09:
Thoughts on this @nfeuer?

nfeuer, 2021-10-04:
@TimoWielink add this to wiki