ClusterDuck-Protocol/ClusterDuck-Protocol #19: Add info on how to troubleshoot

mkinney, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
After you create a MamaDuck, what should you see in Arduino Serial Monitor? (using 115200 baud)
After you create a PapaDuck, what should you see in Arduino Serial Monitor?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Document some "errors"/problems.
Be sure to change the "Z" in the MamaDuck to a realistic value. (I used mac address without colons and it seemed to work ok.) If not, then you may see Authentication errors on PapaDuck serial output. 
If only see "i", check four values in PapaDuck. 

**Describe alternatives you've considered**
Add some troubleshooting/helpful info.

**Additional context**
Show info on what you should see on IBM side.

mkinney, 2020-03-14:
This is what I see on PapaDuck serial output: (with some values obscured)

```
Serial start
LoRa On

Connecting to MYSSID..............................................................<snip>........................................................................................................................................................................................................................................
WiFi connected - PAPA ONLINE
PAPA Online
Reconnecting client to XXX.messaging.internetofthings.ibmcloud.com
136
User ID: 246f28b54624
Message ID: a8j:bkx4
Message: D6ju2LNqIyqq*Mike**None*1234567890*false*false*false*false*false*false*false*false*0*0*0*0*None*
Path: 246f28b54624
{
  "DeviceID": "246f28b54624",
  "MessageID": "a8j:bkx4",
  "Payload": "D6ju2LNqIyqq*Mike**None*1234567890*false*false*false*false*false*false*false*false*0*0*0*0*None*",
  "path": "246f28b54624,Papa"
}Publish ok
{
  "DeviceID": "246f28b54624",
  "MessageID": "a8j:bkx4",
  "Payload": "D6ju2LNqIyqq*Mike**None*1234567890*false*false*false*false*false*false*false*false*0*0*0*0*None*",
  "path": "246f28b54624,Papa"
}
```

mkinney, 2020-03-14:
This is what I see on MamaDuck serial output:

```
Serial start
Created Hotspot
Created local DNS
LoRa On
MamaDuck Online
data received
Mike
Portal data received
wQ2JshY7fp0x*Mike**None*1234567890*false*false*false*false*false*false*false*false*0*0*0*0*None*
Here7
Packet sent
Message Sent
data received
Mike
Portal data received
wQ2JshY7fp0x*Mike**None*1234567890*false*false*false*false*false*false*false*false*0*0*0*0*None*
Here7
Packet sent
Message Sent
```

mkinney, 2020-03-14:
If you visit your dashboard: https://XXX.internetofthings.ibmcloud.com/dashboard/devices/browse. (change XXX to your IBM Org)

For instance: https://ex7n1d.internetofthings.ibmcloud.com/dashboard/devices/browse

Looks like this:
[image]

mkinney, 2020-03-14:
Where should this info be captured/shared?

knouse1344, 2020-03-28:
this is good stuff @mkinney .  Maybe would be best on like an FAQ / help page?  What do you think?  Slack is great for active conversations, but needs to be a place for documented things people can refer to.

mkinney, 2020-03-28:
Sure. A "doc" folder could be created and some markdown could go in there (like "faq.md" or "troubleshooting.md")

Could also use the github wiki functionality. Up to you how you want to do it.

I think a lot of the readme could be moved there, too.

knouse1344, 2020-03-29:
got it.  will get @TimoWielink on this.

krook, 2020-04-14:
Thoughts on this @TimoWielink?

nfeuer, 2020-06-18:
@TimoWielink you should revisit this while you are working on the Wiki

TimoWielink, 2021-04-26:
Add more to FAQ page, with recent bugs and problems we have found.

closed 2021-04-26 by nfeuer