ClusterDuck-Protocol/ClusterDuck-Protocol #51: Make Configurables more configurable

gozu42, 2020-05-11:
**Is your feature request related to a problem? Please describe.**

currently everything is hardcoded and in just two files which can be very annoying when for example european ducks should use different RF parameters.

**Describe the solution you'd like**

i would like to move the currently hardcoded configurables out of the cdp library code.

as step one, just replace them all by CDP_something defines that go into a separate cdp-config.h in the same dir.
while this is not a longterm-best solution, it is one that is really low effort and unlikely to break anything and already solves the "i commited the wrong frequency/pinout by accident" issue.

this also bundles all configurables in one place which is a good step towards making them actualy configurable.

next step could be to move some of them into even more separate .h files to keep pinout specs for different boards or sensor loadouts or deployment-specific wifi setups in one place each.

this also makes it more obvious which parts could be useful to have runtime-configurable. 

**Describe alternatives you've considered**

there is certainly potential to use an "if xml is not solving your problem, you are not using enough of it" approach.
but i dont see any way to make it as low-effort, gradual and "as needed" scalable. 
the define-and-refine way also leaves all options open for runtime-config-storage later, so we can see what works best when we get there.

**Additional context**

candidates, grouped by "dynamic range":

* unlikely to change in a dynamic way
* * hardware details (pinout)
* * hardware presence (do we need to init display, powerchip, vext,...)

* unlikely to ever change within a deployment
* * RF parameters (radio type, frequency, bandwidth, spreadfactor, coding,...)
* * timings (sensor frequency, alivequacks,...)

* more likely to change
* * wifi details (both papaduck-uplink and allducks-AP)
* * password(s) (config page, wifi,...)

this change will be all over the cdp codebase for the first step, so will have maximal merge-conflict potential there. otoh, that first step is 1-3 hours max.
i would like to get an OK for this from @nfeuer and suggest we coordinate a suitable timeslot (where no one else has major changes open) on slack.

after that first step, the potential for merge conflicts is greatly reduced, so any additional steps (like adding the definition for another board) should require almost no coordination.

gozu42, 2020-05-12:
the defines from arduino env for the different lora board types i have here:
```
-DARDUINO_T-Beam -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"T-Beam\"" "-DARDUINO_VARIANT=\"t-beam\"" -DESP32

-DARDUINO_HELTEC_WIFI_LORA_32_V2 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"HELTEC_WIFI_LORA_32_V2\"" "-DARDUINO_VARIANT=\"heltec_wifi_lora_32_V2\"" -DESP32

-DARDUINO_AVR_FEATHER32U4 -DARDUINO_ARCH_AVR 

-DARDUINO_TTGO_LoRa32_V1 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"TTGO_LoRa32_V1\"" "-DARDUINO_VARIANT=\"ttgo-lora32-v1\"" -DESP32
```

this covers most pinout-ifdef needs.
it does _not_ help at figuring out whether the given t-beam has a axp192 or not.

gozu42, 2020-06-01:
segmenting this into iterations that are "sprint sized" and result in a "working PR" after each round:

* replace hardcoded parts by defines ... see https://github.com/Code-and-Response/ClusterDuck-Protocol/pull/71
* add https://github.com/Code-and-Response/ClusterDuck-Protocol/issues/52 so users can disable debug chatter on the serial (without removing all the debugprints we might need again the next time we have a memory leak hunt)
* add an include-if-present cdpcfg-user.h (so users can override f.ex.RF LORA params)
* define board-specific blocks ... for https://github.com/Code-and-Response/ClusterDuck-Protocol/issues/62 and/or https://github.com/Code-and-Response/ClusterDuck-Protocol/issues/68
* turn parts that should be runtime-configurable into nvs-backed config that defaults to the defined values

nfeuer, 2021-01-11:
OOP has been implemented extensively as well as expanded our config file. This makes it a lot easier to make things more configurable. Some of these items have not been completed yet but should be broken out into their own issues.

closed 2021-01-11 by nfeuer