ClusterDuck-Protocol/ClusterDuck-Protocol #52: improved logging

gozu42, 2020-05-11:
**Is your feature request related to a problem? Please describe.**
currently the logging is just Serial.print all over the place which makes it hard to focus on individual areas when needed.

**Describe the solution you'd like**
i would like to introduce a log function that is similar to the classic unix syslog facility+severity concept.

so each of these serial.print gets classified by "what is this about" (facility, like "lora" or "parser" or "web" or "uplink"...) and "how important is this" (severity, like "debug", "info", "warning", "error", "kaboom"...).

and the developer/user can decide what they want to see, like "web at full debug, everything else warning and above".

the simple way to do this is a lot of defines.

**Describe alternatives you've considered**
i looked at a few arduino logging frameworks, but they were all either lacking in relevant functionality or were really heavy-handed.
and adding yet-another-external-dependency for a developer-only feature seems bad.

the define approach has the advantage that for the simple case it is compile-time evaluated (meaning disabled messages dont are not even in the compiled binary).
and at the same time it allows to retrofit anything under the hood without touching the main code again, so if someone wants their papa to log to mqtt, no need to change the lib.

**Additional context**
this has maximum merge-conflict potential since it needs to touch every single Serial.print currently in the code.
the concept/idea can be demonstrated as part of a out-of-tree special-purpose-duck, then do the main change as a compact 1-3 hour session.
i suggest i do the demonstrator, get feedback/approval from people, then coordinate a change-window via slack.

gozu42, 2020-06-18:
while trying to figure out platformio on the t-beam, i "found" the way the esp32 core handles debug logging.

example message on the serial console:
```
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
```
that is pretty cool/verbose since it tells you _all_ the possibly relevant metadata of the message too: severity (D)ebug, sourcefile:line, function() ... 

the line of code that generated it is very simple though:
```
    log_d("PSRAM enabled");
```

so the whole prefix "[D][esp32-hal-psram.c:47] psramInit():" is actualy generated by some strategic define magic: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-log.h

in the code, you just use it like this:
```
  log_n("logtest n");
  log_e("logtest e");
  log_w("logtest w");
  log_i("logtest i");
  log_d("logtest d");
  log_v("logtest v");
```

and it is  with printf syntax, so it could be log_n("the number: %u", number) too.

and while building, you can select which of these show up.
in my platformio.ini i added:
```
build_flags = -DCORE_DEBUG_LEVEL=3
```

this makes the n+e+w+i levels show up, but not the d+v
```
1592480691  gSS: 286208 gFSS: 1310720 gSM: b5b8eac675ec8bfdfe55147e301f00a2
-- begin of the test block --
1592480691 [E][RAMster.ino:25] setup(): logtest n
1592480691 [E][RAMster.ino:26] setup(): logtest e
1592480691 [W][RAMster.ino:27] setup(): logtest w
1592480691 [I][RAMster.ino:28] setup(): logtest i
-- end of the test block --
1592480691 [E][RAMster.ino:32] setup(): setup done
```

if i want to see d+v too, then i need to set it to =5.

the default debug level is 1, so it shows (e)rrors and (n)one level.

### pros:
* readily available and easy to use
* compile time, so messages that are not shown are not even in the binary

### cons:
* esp32 specific (though single .h file that could be easily ported/replaced) 
* compile time, so no runtime adjustment of loglevel possible (we dont have that right now either)
* no "facility" functionality, so you can adjust just one global debug level (we dont have that right now either)

### steps:
basicly the same as in the initial comment:
this has maximum merge-conflict potential since it needs to touch every single Serial.print currently in the code.
i would do the change if i get some feedback from the usual suspects and a go-ahead for a 1-3 hour change window where no one else has relevant unmerged changes.  
ping @nfeuer @TimoWielink

nfeuer, 2020-06-18:
I think this looks good. I will review the PaPi branch right now and try to merge it into master. Once that is done you should have your window.

nfeuer, 2021-01-11:
Logging has been expanded so you can see which file and function its coming from as well as more extensive error handling. We have not implemented a defined approach yet for turning parts on and off yet. We can reopen a new issue to address.

closed 2021-01-11 by nfeuer