ClusterDuck-Protocol/ClusterDuck-Protocol #53: improved duck telemetry

gozu42, 2020-05-11:
**Is your feature request related to a problem? Please describe.**
currently the alive-quacks do not have any real information content.
there also is no real "monitoring of duck health" beyond it.

**Describe the solution you'd like**
i would like to add some telemetry info to the alive quacks.

* basic - device independent infos (only depends on arduino framework)
* * uptime (millis())
* * loopcount (have a counter at the beginning of loop(), this is a simple way to measure how "busy" an arduino device is)
* * memory stats (free mem, max block, low water mark,...)

* advanced - device specific infos (needs to be part of a board definition)
* * battery level (adc pin)
* * power stats (axp192 on the newer t-beams)

these could all be added in simple string format as space-separated tag=value pairs to the alive message.

**Describe alternatives you've considered**
this could be done in a separate "telemetry quack" message, but i do not see the advantage over just sending it as part of / replacement for the alive-quack.
the text-format is not important to me, if there is an advantage for dms/backend in making the telemtry json right away, just holler.
something like the loopcount is "more natural" to do in the duck-app-sketch, but can also be contained inside the duck-class.

**Additional context**
the basic infos have very low merge-conflict potential since it mainly just changes the alive function.
i would like some basic feedback on format and suggestions for other infos to include.
advanced infos can be added later once it is more obivous how/which boards are supported/used.

Evilgeniusnerd, 2021-04-17:
@tqrahman  maybe should update this with your ongoing efforts?

tqrahman, 2021-04-20:
yeah so far we have GPS, Altitude, Board Temperature, Volts, isCharging for the TBeam.

nfeuer, 2021-04-26:
This has been completed, just need to add memory free

nfeuer, 2021-04-29:
With the recent merge we were able to provide almost half of these data points

nfeuer, 2021-10-04:
Created new issue #241

closed 2021-10-04 by nfeuer