ClusterDuck-Protocol/ClusterDuck-Protocol #54: quickstart example

gozu42, 2020-05-11:
**Is your feature request related to a problem? Please describe.**
i would like to have an exampleduck that makes it really easy to get started with _something_.
currently there are waaaaaay too many steps/decisions required for a "first quack".

**Describe the solution you'd like**
first goal would be to have an exampleduck that can be compiled as-is.

* a mamaduck,
* using last 16 bit of mac-addr as duckid (4 chars),
* with a statuspage on the webserver showing the last N quacks received

perhaps it is even possible to get rid of having to define a board-type by looking at what the arduino framework defines for this.
the esp32 has enough ram that adding this "status page" should be doable, there should be no impact on builds with this feature disabled.

once the basics are in place this can easily be extended to add a basic "lora CB-like chat".

**Describe alternatives you've considered**
i dont see any alternatives since no amount of documentation can replace a zeroconf experience.

**Additional context**
the merge-conflict potential seems low since this is mostly an additional example plus an added page in the webserver.
i did not look at the webserver code close enough yet, if it looks problematic i will open another enhancement issue for splitting the captive-portal code from
the protocol code.

Salzi, 2020-05-11:
A slightly different discussion would be: Do we really need so much steps/decisions/variants? To be honest, I never really understood why it needs three different types of ducks. Wouldn't it be much easier if all ducks could just do the same thing? I don't know if this would work with Papa functionality, but ducklinks have no advantage over mama ducks.

knouse1344, 2020-05-11:
is true.  i've thought for some time to merge Mamas and DuckLinks.

krook, 2020-05-11:
I think it would be good to diagram out a migration plan proposal and work through a comments phase (maybe even just a community call to hash out the pros and cons and come to a next step). 

That way we can agree on a good approach without doing the code-first approach (I did like that PoC approach in the WIP PR) that may not be cleanly mergable.

gozu42, 2020-05-11:
lets move the code-structural discussion over to #56

nfeuer, 2021-01-11:
We have implemented something similar. Does not use the MAC address for duck ID, but is a simple first-time setup. Once set up it will ping a message counter. You can set up two devices to send messages back and forth to each other. 

https://github.com/Call-for-Code/ClusterDuck-Protocol/blob/master/examples/1.Ducks/MamaDuck/MamaDuck.ino

nfeuer, 2021-01-11:
We should make sure this is highlighted in our getting started documentation. It might be a bit harder to find with the current file setup

closed 2021-01-11 by nfeuer