ClusterDuck-Protocol/ClusterDuck-Protocol #55: Build out contribution model

krook, 2020-05-11:
Create a system that leverages GitHub projects, wiki, milestones, and issues in order to clearly show open bugs and future directions. This will support the community by showing cutting edge work, help needed, and release schedule.

gozu42, 2020-05-11:
(i am focusing a bit on the technical parts)
specifying a "branching model" should be part of this.
like ... 
* c-a-r master is HEAD, this is the official ground-truth future.
* all code enters car-master through pullrequests.
* p-o creates "deployment branches" (in the po repo) off that as needed, so the "volcano 237" mesh has a stable codebase that is still versioned properly and can cherry-pick bugfixes off car-master as needed.
* there is no real need to do development in branches on the c-a-r repo as long as people push their intermediate results to their own repo. there is a nice graphical view on the branches across all forked repos (use drag and mouseover!): https://github.com/Code-and-Response/ClusterDuck-Protocol/network
* this works better if the development happens on branches even if in a private/forked repo because then you can pull in official car-master to master and rebase your changes as needed locally before push+PR.

this should work without giving "random externals from the internet" (like me) actual permissions to anything, and can still be used with an informal "if two people on this list approved and CI is green, its ok to just click merge" rule. 
(perhaps this can even be automated with github actions? @Salzi ping )

krook, 2020-05-13:
First draft of roadmap posted in this doc: https://docs.google.com/presentation/d/1_fhRhRLJmVwc8d9Ra_q4qP2CPFjPWiy6TKfd1o_Oo2E/edit#slide=id.g803e77a99e_5_6

krook, 2021-10-13:
Looks like this can be closed. Great work on documentation in the wiki.

closed 2021-10-13 by krook