ClusterDuck-Protocol/ClusterDuck-Protocol #56: Adopt an object-oriented approach for ducks

krook, 2020-05-11:
Based on the approach started by @Salzi in #37 

The goal is to reduce the size of the codebase, eliminated duplication, and reduce the surface area for potential bugs. This can also help reduce the number of duck types (DuckLink vs Mama).

gozu42, 2020-05-11:
my impression was that some of the duplicate code was due to "all configuration is handled at compile time". 
so there is some tie-in with later stages of #51 because with runtime-configuration a ducklink is just a mama that is set to not do lora-receive.
meaning ... i would prefer not to replace "everything is hardcoded" by "everything is hardcoded with less duplicate code thanks to inheritance". 

so to me it looks like the base-class is close to "mama", with config-disabled lora-listen you have a ducklink, and with oo-inherit-overloaded send-quack function you get a papa.

not sure how much sense this makes. because how do we treat other "optional code", like the sat-modem, or optional sensors? 
perhaps a "has a" pattern makes more sense than an "is a" pattern?

then the central class is just a "duck", and it has a variable amount of quack-sources (captive-portal page, sensor task, lora listen, bluetooth,...) and a variable amount of quack-targets (lora send, mqtt, serial, logpage,...)

then we have a pretty direct mapping for the current duck types:
ducklink: src(web) + trg(lora)
mamaduck: src(web,lora) + trg(lora)
papaduck: src(web,lora) + trg(dms)

sources need 
a setup function that gets passed a ref to their duck, 
and a loop function.
they call a pleasehandlethisquack function on their duck as needed.

targets need
a setup function that gets passed a ref to their duck, 
a loop function.
and a sink function that gets called from the duck with quacks to send. 

this needs input from someone who actualy understands OO and CPP and perhaps the arduino env.
@Salzi ?

nfeuer, 2020-11-30:
This has been addressed in the new refactored code with the latest update

closed 2020-11-30 by nfeuer