ClusterDuck-Protocol/ClusterDuck-Protocol #58: Send data over RockBlock Instead of WiFi - Iridium Satellite Network

TimoWielink, 2020-05-20:
**Support RockBlok**
Iridium Satellite Communication

The goal is to connect the Papa Duck to the Iridium Satellite Network, to send data from places where there is no LTE or WiFi Available.  

We want to integrate the Rockblock module to the different ESP32 boards and add the IRIDIUMSBD library for the RockBlock Module.

Currently you can only send data from the papa straight to an MQTT server by WiFi, so adding the Rockblock we would be able to send data an alternative route. 

- [ ] Setup RockBlock module

- [ ] When a message comes in we need to convert to a  const char so we can send with RockBlock

- [ ] When we converted message we can send the message to the network

- [ ] Write documentation for future wiki, about tokens/use of iridium

TimoWielink, 2020-05-21:
The first example file for the RockBlock is in the #60 Branch: 58-DishDuck-Iridium. This file works when a message comes in over LoRa it triggers the Rockblock send and sends a Char yo the iridium network with the Duck message.

Evilgeniusnerd, 2020-05-22:
current issue is once you send the first payload of data. It does not repeat. The code breaks because the RockBlock Modem needs some time after transmissions is my guess. Have yet to figure out why it crashes.

Evilgeniusnerd, 2020-05-22:
Also a detailed [README.MD](https://github.com/Evilgeniusnerd/ClusterDuck-Protocol/blob/58-DishDuck-iridium/examples/PAPA_IRIDIUM/README.md) is available for anyone who wants to know how to use the RockBlocks with the TTgo T-Beam and Heltec ESP32 Lora Boards.

closed 2020-06-18 by nfeuer