ClusterDuck-Protocol/ClusterDuck-Protocol #65: Add Encryption

nfeuer, 2020-05-29:
Under normal testing purposes, protecting data over the CDP network is not an issue. However, in a natural disaster situation, personal information might be required to be shared for emergency personnel to respond to the situation. Therefore, the CDP needs the capability to protect the information until it reaches the first responders. This is also important for protecting information sent over different networks. 

There are some challenges for security as a distributed network. How do we prevent someone from spoofing a Duck on a secured network? How do we prevent malicious data being transmitted over the network? How do we recognize Ducks in the same “family”? 

Some approaches we have looked into in the past:
- 256-bit encryption
- Byte-codes (like a code/sync word)

nfeuer, 2020-08-17:
@tqrahman bump update

tqrahman, 2021-02-11:
Encrypting using the [Arduino Cryptography Library](http://rweather.github.io/arduinolibs/crypto.html).

Evilgeniusnerd, 2021-04-17:
@nfeuer  @tqrahman status update on this?

tqrahman, 2021-04-20:
we already implemented AES256 encryption with a CTR mode. It is merged with master already. checkout DuckCrypto.cpp. The next step is to create a documentation for it.

closed 2021-04-26 by nfeuer