ClusterDuck-Protocol/ClusterDuck-Protocol #67: Fix Network Pathing/Hopping

nfeuer, 2020-05-29:
Each time a Duck sends a message, it adds its own digital signature to the “path” portion of the message. Since each message is broadcasted to all nearby devices, a Duck will receive a message it has already sent. To prevent the same message from being sent infinitely in a loop, the Duck will check the “path” section for its digital signature using `idInPath()`. If the Duck id is in the "path", it will ignore the message. This has worked well in linear and smaller-scale networks. However, this becomes an issue in large scale networks with many devices and long device ids. Each LoRa message cannot exceed 256 bytes per packet. While splitting packets could work to extend the message size, it increases device busy time and network traffic.

**Proposed solution**
One idea to fix this is by assigning each Duck with a value dependant on how far away it is from the PapaDuck. If receiving a message from a Duck with a higher value than its own assigned or self-assigned value it won't send the message.

Here is a paper that touches on the idea:
[ProximAid: Proximal adhoc networking to aid emergency response](https://ieeexplore.ieee.org/document/7402031) 
Message me on Slack if you are interested in learning more about this paper.

nfeuer, 2020-06-18:
Working on a new idea. Using a Bloom Filter to get a degree of confidence that a message hasn't been sent yet. A quick intro to get an idea of how it works: https://www.youtube.com/watch?v=gBygn3cVP80

Working on a test implementation

gozu42, 2020-06-18:
some notes on this.

i suggest to play around with the numbers in some online BF calc, this one seems to work:
https://hur.st/bloomfilter/?n=&p=1.0E-6&m=64ki&k=16

these numbers are for a 8kByte sized filter (== 64kBit == 16 bit sized keys), with 16 keys per message. at 1:1mil false-positive chance this means a filter is good for 2244 messages.
if you change the probability to 1:10mil it is still 1862 messages.
or the other way around, at that filter size for 2000 messages you get a 1:4mil chance.

the 16-bit size and 16 keys is based on.... how do we generate them?
simple way: take sha256(salt+message), thats 16x16 bits output.
"message" is the non-variable parts of the quack, currently thats msgid+senderid+payload bytecodes (but not the path!).
"salt" is a random value (256bit size if you dont want to think about it) the duck makes up when it creates the filter (and remembers for the duration of using that filter!), this is basicly "free" and means that even if the 1:4mil chance hits, it is not going to hit on two ducks at the same time!
(so if your mesh is an actual mesh, the very unlucky message will still get through. the chance that a message is so mega-unlucky to be unlucky on _all_ involved ducks at the same time is ... astronomic.)
if we need more than 256b hashbits we can use either sha512 or do sha256-of-sha256 (which is super-ok for BFs).

since BFs dont allow _removing_ entries (yes, there are "counting BFs", those add a lot of complexity and you still need to handle the case where the bucket runs over, so not much won at a steep price), the simple way is to ping-pong between two filters.
* so we start with filter A (which has a counter about how many messages got added and a salt), 
* * and keep adding+checking, 
* and when the counter hits 2000, 
* * we switch A to readonly "check but dont add", 
* * create a new filter B (new salt, new counter) and 
* do check-A and check-and-add-B for the next 2000 messages,
* * set B to readonly "check but dont add",
* * throw away old filter A and 
* * replace it by a new empty filter A with a new salt and empty counter,
* run as check-B and check-and-add-A for the next 2000
* ... 
you get the idea.

that way we dont have a precise amount of last-messages-to-check but between 2k and 4k.
(except while we wait for the initial filter "A" to fill, during that window it is less than 2k)
if A and B have a different salt we need to do two sha256() ops, which doesnt matter since the esp32 can handle more than 10k sha256 per second. 

there are BF libraries for arduino, but basicly this is so easy to implement that i am not sure using a lib is worth it. 
the libs are ok if you want dynamic sizing or more advanced feature sets, but basicly here we only need a simple filter with "create", "check" and "add"...

gozu42, 2020-06-22:
i slapped together a from-scratch minimalist (<100 SLOC) BloomFilter implementation: https://github.com/gozu42/BloomTest

the dummy sketch does statistical tests and the numbers are as expected (it finds all added keys (== no false negatives) and has about as many false positives as math says we should have).

one thing missing in the above numbers: the 4mil for 16x16 @2000 is ... for one filter. so with the A+B setup, we are moving between 1:4M (right after we swapped filters, so one is full and one is empty) and 1:2M (right before we swap filters, so both are full).

whats missing is a design decision on how to integrate this into CDP, i am trending towards just putting it into a cdp-bloom.h and including it deom CDP.cpp. spinning it off into an actual library seems overkill for 100 lines of code, but integrating it into the Duck class seems wrong too because it _is_ somewhat separate/contained functionality.
there are also "if not sure because no strong reasons to go X or Y, do the variant that is less messy to undo" considerations at work. 

design arguments welcome, if there are no major complaints/suggestions i will go ahead with the integration as .h in a WIP branch in the next days.
i intend to keep "pathing" in and have CDPCFG_FEAT_PATHING and CDPCFG_FEAT_BLOOM toggles to enable either, or both (!) with a "compare results and complain loudly if they mismatch" message.

amirna2, 2024-03-06:
This thread is no longer relevant given PATH tracker is no longer part of CDP

closed 2024-03-06 by amirna2