ClusterDuck-Protocol/ClusterDuck-Protocol #68: Sensor Integrations

knouse1344, 2020-06-01:
The ClusterDuck Protocol provides a foundation that can enable a variety of sensor applications for different purposes.  An example of this are the SolarDucks in Puerto Rico - many of these have BMP180 (temperature, pressure, altimeter) sensors or an Air Quality PM2.5 sensor.  Most sensors that can be integrated into Arduino can be integrated into a DuckLink.

[image]

**This issue focuses on two main goals**
1. Constructing the process for future integrations of sensors
2. Building the integration of numerous common sensors already used.  There is already an example for the BMP180 sensor.  Others we are looking to build out
- MQXX Gas sensors (MQ-1, MQ-2, MQ-3, etc).
- Accelerometers
- Air Quality PM2.5 sensors
- Infrared heath thermometers
- Motion/visual sensors

gozu42, 2020-06-22:
i am actualy mildly annoyed by the current situation of this.
the way the various sensor/transport libs are integrated (as libdeps of CDP instead of libdeps of the sketches/ducksamples using the sensors) means every single of my duck builds fetch/install all these libs _and_ their dependencies even though i am not using any of it.
this is more noticeable when doing pio builds for multiple boards since by default it means the problem gets multiplied by board type. 

i am going to look into how to move the libdeps into their examples/blah dirs every time i am waiting for the build being slow.

nfeuer, 2021-02-15:
Should look into having a different library setup for sensors so we don't compile them if they are not used

nfeuer, 2021-05-06:
I think I will close this for now since we have added new sensors

closed 2021-05-06 by nfeuer