ClusterDuck-Protocol/ClusterDuck-Protocol #73: Over the Air (OTA) Updates

nfeuer, 2020-06-03:
As the CDP evolves, it will be necessary to update already deployed devices with the new firmware. Currently, the only way to update devices is to physically plug the device into a computer and upload the new firmware. This is a very arduous process as existing devices are placed far apart and are in difficult to reach areas. 

OTA would solve a lot of these issues by allowing devices to be updated without needing to physically interact with each individual device. This would save a lot of time and even allow for networks to be updated remotely.

OTA updating methods:

- [ ]  WiFi

- [ ] Bluetooth

- [ ] Captive Portal

The most challenging, but most valuable:

- [ ] LoRa

OTA updating over LoRa would allow for the whole network to be upgraded at the same time. This is especially useful for large scale deployments and devices in very hard to reach areas.

TimoWielink, 2020-06-05:
With @Evilgeniusnerd we build an extra page into the AsyncWebServer where we can reflash the ducks Firmware.

Currently, in the 73 OTA branch there is now a feature where you can upload a .bin file by going to 
192.168.1.1/update and it will reflash the duck.

For this feature to work you will need to directly connect to the duck's Acces point.

TimoWielink, 2020-06-09:
I added a Password and Username protection for the /update URL in #76 so only the owner of the ducks is able to reflash the ducks

gozu42, 2020-06-18:
OTA via captive portal, the textmode adventure.
this works from both regular linux and a raspi.

the wifi client adapter used in my case is a
```
Bus 001 Device 008: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
```
very tiny/cheap usb plug, supported by linux out of the box.
it should work with any other wifi interface too.

basic interface configuration, needs to be done only once:
```
ifconfig wlan0 192.168.1.123/24 up
```

scan for ducks within range:
```
iwlist wlan0 scan | egrep -i "cell|essid" | grep -B1 DUCK
```

connect to a specific duck by mac-addr and optional verify connection:
```
iwconfig wlan0 ap 12:34:5678:90:AB
iwconfig wlan0 essid ' 🆘 DUCK EMERGENCY PORTAL'
ping -c3 192.168.1.1
```

upload new firmware:
```
curl -v -F inobinfile=@src/TelemetryDuck-pio/.pio/build/ttgo-lora32-v1/firmware.bin http://username:password@192.168.1.1/update
```

i have half-working "update all ducks in range" automation here but i am afraid it is perl again and it still needs to learn about different duck-hardware.

gozu42, 2020-06-22:
getting stuck up on pesky details for this.
like there is no general way to figure out what pio-env is used for building. (that means a duck can not tell us what family of image it wants for its update, so we need to keep track of that somehow else)

the appimage header exists and contains app name and version and build date, but in case of pio contains these data for for _framework_ used, not the actual sketch.

i am looking into what apis are provided by the esp-idf for dealing with OTA and the ota partitions in general.

functionality i am looking to implement:
* status page on captive portal 
* * that shows current status (inkl appimage hashes) of the OTA partitions 
* * and which of them is currently running
* * and allows to select which to boot next.
* better HTTP handling
* * currently it just reboots at the end of a successful update, 
* * which means the client is going to time-out eventually, but doesnt get any feedback
* * so the client cant tell if the update actualy worked or if something explode-rebooted.
* look into https options
* * user/pass is good, but currently in plain over plain, so very .. plain.

ota-over-lora seems quite possible as a longer-term option, but still needs a better understanding of what is actualy going on, including content/status of the ota partitions.

some reference links:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html

TimoWielink, 2020-06-22:
I will be able to support these new development features.  My first step will be providing feedback on the update itself, we need to apply error handling and a safe reboot from the previous firmware.

nfeuer, 2021-01-12:
Breaking out issue into smaller issues

closed 2021-01-12 by nfeuer