ClusterDuck-Protocol/ClusterDuck-Protocol #85: Build a buffer for incoming mesages for PAPA

TimoWielink, 2020-06-18:
**Is your feature request related to a problem? Please describe.**
When a message comes into the papaDuck but the papaDuck is trying to connect to the internet the incoming messages get lost.

**Describe the solution you'd like**
We need to build some sort of storage/queue where the incoming messages can sit temporarily 

**Additional context**
When I was developing for the DishDuck I came across a problem where it takes a little while before the RockBlock is booted so the incoming messages get lost during that time.

gozu42, 2020-06-22:
this should be applied to MAMA too, radiolib now provides an interface for "is the lora modem receiving right now" and by checking that before sending via lora we can avoid at least some on-air
collisions.

a first step in this direction would be to define a struct (or even class) that contains everything related to a message/quack.
at the moment all data fields are spread over the Duck class, and functions have interesting assumptions about which fields are valid and contain what at any given time.

there may be some possible tie-in with #88 because one way to go about this could be to keep the data as key-value pairs and using the bytecodes as keys. that way we could avoid translating between "names" and "codes" multiple times during the lifetime of a message.
as in, hold the message in a base format, and have a bunch of "send as lora" or "send as json via mqtt or http" or whatver sinks.

nfeuer, 2020-08-17:
This should be tied in with #99

closed 2021-04-29 by nfeuer