ClusterDuck-Protocol/ClusterDuck-Protocol #102: Get more feedback from AXP192 when using T-Beam or other AXP enabled devices

TimoWielink, 2020-07-06:
**Is your feature request related to a problem? Please describe.**
With the build in AXP192 we can get data like Voltages, Watts, Is charging, battery percentage, etc. Which can be very useful to see when the ducks are deployed in remote areas.

**Describe the solution you'd like**
Integrate AXP Library to start integrating the Pre-build methods for retrieving power data 

**Additional context**
V1 is developed and I will make a PR ASAP.

TimoWielink, 2020-07-20:
Created PR, having some problems where it wont Publish all of my power data only when its a small "string", might be a problem with the arduinojson?

#113

nfeuer, 2020-08-17:
MQTT issue solved. Just need to make pull request.

closed 2020-11-30 by nfeuer