ClusterDuck-Protocol/ClusterDuck-Protocol #105: Payload as JSON

nfeuer, 2020-07-09:
To make data analysis easier, serializing data as JSON would make it easier to transform the data on the cloud side. However, for the captive portal that has many fields, this can use up a lot of the limited number of bytes that can be sent over LoRa. Therefore, the solution needs to simplify the JSON document as much as possible so that we have the most available bytes for emergency messages/data. 

One idea was to have a code in the root that would correspond to a template stored on the cloud. While this could work well in simplifying the JSON doc, it increases the complexity needed on the cloud side.

BrentonPoke, 2020-08-17:
Just as a thought, one alternative to json is [MessagePack](msgpack.org). It's basically a condensed version of json that works at the byte level with small ints and schema info stored as a single byte. The document ends up being smaller than json overall, and there's a client library for basically every language.

knouse1344, 2020-09-11:
so this is really cool https://msgpack.org/.  You can "try" it and just paste a json string.  so i took an example transmission from CDP:
`{"DeviceID":"HiMyNameIs", "MessageID":"2jy5ep:y", "Payload":"Current dust concentration: 11.58", `"path":",HiMyNameIs,PapaDuck"}`

msgpack takes it from 127 to 109 bytes.  This is meaningful, and we should seriously consider implementing it, but at the same time its not an order of magnitude or so greater so im thinking this should be paired with some sort of network architecture management.  I think a similar wireless lora radio company "limits" hops on the devices to 6 hops, more or less preventing an overload of large meshes.  This gives more opportunity for multi-packet.  Compression is very useful here.  But maybe, is an absolute commitment to single-packet transmissions dogma or just something that is useful to a point?

BrentonPoke, 2020-09-11:
Compression is always an option. [Snappy](https://github.com/google/snappy) is one library with licensed under BSD that aims for speed and moderate compression, while [Boost has an api](https://www.boost.org/doc/libs/1_72_0/libs/iostreams/doc/classes/bzip2.html) for bzip2 that makes it easier to use. bzip2 will likely have much greater compression, but not be as fast.

faradaym, 2024-08-26:
We have no plans to enforce a data format at this time

closed 2024-08-26 by faradaym