ClusterDuck-Protocol/ClusterDuck-Protocol #106: Enhance the beginner experience with simple steps on getting started

krook, 2020-07-13:
**Overview**
As more people join the community, it's good to be able to get them engaged as soon as possible. Whether that's by adding a demo video or overview showing how things work, or by creating short tutorials with simulated pieces (like using Paho to send MQTT messages that are later shown on their own DMS Lite, or by showing how to receive and interpret a message.

**Additional context**
After Call for Code submissions end on July 31, we'd like to provide IBM Developer visitors to go explore and contribute to previous projects that are now open source. They should be able to use the tech, or clearly see how they might open an issue or submit a pull request for basic improvements.

TimoWielink, 2020-07-20:
We are currently working hard on rewriting the Quick Start to get started with the CDP. I and @knouse1344 are working on an instruction video that should be a good addition to how to build a Duck.

nfeuer, 2020-07-20:
This could be an option? 

[ESP32 Simulator](https://github.com/mireq/esp32-simulator)

krook, 2020-07-29:
Would love your feedback on what can be done to improve the experience @Evilgeniusnerd @trentbrown13 and others.

nfeuer, 2020-08-17:
Maybe we can make some beginner projects? Build your own weather station?

nfeuer, 2020-08-17:
- Quick start guide video of a powerpoint presentation
- Make a list of project ideas available on the website
- More youtube videos

Evilgeniusnerd, 2020-09-05:
We have two scripts and documentation covering this one https://github.com/amirna2/ClusterDuck-Docs/blob/master/cdp-install-osx.sh. and another https://github.com/Code-and-Response/ClusterDuck-Protocol/wiki/Linux-CDP-Dev-Setup

knouse1344, 2020-09-11:
videos drafted, will share videos in next couple days

knouse1344, 2020-11-20:
working thru an improved variant of this

krook, 2020-12-04:
Looks like there's been a lot of great documentation added to the wiki!

closed 2020-12-04 by krook