ClusterDuck-Protocol/ClusterDuck-Protocol #107: DishDuck Hybrid Inconsistency Issues

Evilgeniusnerd, 2020-07-14:
**Describe the bug**
I have been doing testing on the DishDuck Hybrid this past week. I have found multiple issues as a result of this testing that needs to be fixed.  First is the "Hybrid Portion" when you are out of range of WiFi Rockblock is suppose to send the messages, Well I tested this and while it does work it is really inconsistent. Meaning that it does not always switch from one to the other. The second issue is that if the DishDuck is flashed without an SSID and Password, You should be able to goto PAPA Duck WiFi setup and set one. But that does not work at all as of right now get a watchdog error and TTgo T-Beam just reboots. Also (SSID Available Scan and Connect) hangs pretty often not allowing the T-Beam to carry on with operations.  

This is the WatchDog error when trying to use PAPA WiFi Setup Page
[image]

knouse1344, 2020-09-11:
so it appears after network scan or network(s) identification.  must be some configuration switching back and forth?  potentially serial turns off other components?  Amir's new PR [here](https://github.com/Code-and-Response/ClusterDuck-Protocol/pull/131) helps to show how we could more simply switch components (WiFi to serial to satcom, or something like that).  @Evilgeniusnerd let's play with this next couple days?

amirna2, 2020-09-11:
Do we think this is an setup/configuration issue or some logic problem in the CDP Protocol?
The DuckBuilder is only helpful in the setup part of the duck and doesn't affect any implementation logic.
I'd like to get a bit more familiar with this `DishDuck Hybrid` duck.

knouse1344, 2020-09-11:
We're not sure yet.  we can narrow down with more testing.  It happens when the DishDuck detects it has lost wifi network connectivity, then it automatically switches to satellite connection.  When attempting to switch back the error occurs.  might be some configuring or something more fundamental to the firmware/electronics.

TimoWielink, 2020-11-16:
Seems to be working better with latest changes on Duck-Components

Evilgeniusnerd, 2020-11-16:
Yes, this can be closed out now. It has successfully been switching between WiFi and Sat-Comm perfectly without any issue.

closed 2020-11-16 by Evilgeniusnerd