ClusterDuck-Protocol/ClusterDuck-Protocol #111: Add support for WS2812 LED

TimoWielink, 2020-07-18:
I would like to give feedback from the Ducks Health by changing the color of a LED.

For example, When the battery is almost dead (get info from AXP202x Chip on the Te beam) we turn the LED red and when its fully charged Green.

- [ ] Make example code for simple LED

- [ ] add support for FastLED Library.
To run different WS2812 LED chips we need to add FastLED library support.

- [ ] Intergrate with AXP202X Lbrary for changing colors

TimoWielink, 2020-07-18:
What i noticed in my testing is that the 5V pin on the TTGO always provides power to the LED even when the board is off, we need to find a solution to power down the led (Maybe Mosfet?)

TimoWielink, 2020-07-20:
Simple implementation of powering and setting up an LED #112

nfeuer, 2020-08-17:
Merge PR #112 and #113. The support works for Heltec, but has issues with TTGO atm

nfeuer, 2020-08-20:
#112 and #113 have been merged

closed 2020-08-20 by nfeuer