ClusterDuck-Protocol/ClusterDuck-Protocol #114: Initial bringup of DuckLinkExample and DuckDetector, Issues noted:

dmandala, 2020-07-29:
Noted two different things:

1) Was told that the DuckDetector should NOT have the web server portal running nor should the emergency page be available.   

2) With the DuckLinkExample code running on a Heltec LoRa 32 and the DuckDetector code running on another Heltec LoRa 32 the DuckLinkExample is never seen.  The two units are about 3 feet apart.

When attached to the SerialPort of the DuckDetectorand reset is pressed I see:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Serial start 115200
Starting LoRa......
LoRa online, Quack!
Listening for quacks
Created Wifi Access Point
Created local DNS
Setting up Web Server
Detector Online
Ping!LORA SND millis:4068 size:4 data:f4023000
startTransmit Packet sent
Ping!LORA SND millis:7068 size:4 data:f4023000
startTransmit Packet sent

... endless repeat ...

Ping!LORA SND millis:10068 size:4 data:f4023000
startTransmit Packet sent
Ping!LORA SND millis:13068 size:4 data:f4023000
startTransmit Packet sent
Ping!LORA SND millis:16068 size:4 data:f4023000
startTransmit Packet sent

I would expect to see:

Ping! Pong! from the DuckLinkExample running.  It never happens.

Evilgeniusnerd, 2020-07-29:
[https://github.com/Code-and-Response/ClusterDuck-Protocol/tree/master/examples/DuckDetector](url)

dmandala, 2020-07-30:
Recompiled using the code at the DuckDetector link above.  With the DuckLinkExample device unplugged, the code still exports a WiFi  connection and that connection still has a web server running and a portal page on it.  Once I plug in the DuckLinkExample device the DuckDetector device still does not see  DuckLinkExample device.  I also have a TTGO device I'll try swapping it in and seeing if I can get a DuckDetector to see it.

nfeuer, 2020-08-17:
We need to take out the `duck.setupAp(false)` and that should stop the web page from popping up.

nfeuer, 2020-08-18:
@dmandala can you try the fix in the most recent PR? See if it solves your issue

nfeuer, 2020-08-20:
Tested the latest fix. There was an issue with the 'setupWifiAP(false)` that was causing the issue. Will need to fix that in another issue.

closed 2020-08-20 by nfeuer