ClusterDuck-Protocol/ClusterDuck-Protocol #116: MaMaDuck example captive portal form not publishing when fully filled out

trentbrown13, 2020-08-05:
**Describe the bug**
while testing, I noticed the captive portal would fail to publish (PaPaDuck => MQTT) when it was fully or mostly filled out.  It will fail when I just add my full name and address but succeed if I just have a few char's in multiple fields.

This smells like a "size" or number of char issues - I'll try to narrow it down

Working publish with just a few random chars entered in the form

16:19:32.879 -> handlePacket pSize 47
16:19:32.915 -> LORA RCV millis:94378 rssi:-47.00 snr:6.25 fe:11618.22 size:47 data:f5025200f609793933787572396c00f711742a2a732a7364662a2a2a2a2a2a2a2a00f3025200e30773746174757300
16:19:32.915 -> getPacketData senderId_B Len = 2
16:19:32.915 -> getPacketData User ID: R
16:19:32.915 -> getPacketData messageId_B Len = 9
16:19:32.915 -> getPacketData Message ID: y93xur9l
16:19:32.915 -> getPacketData payload_B Len = 17
16:19:32.915 -> getPacketData Message: t**s*sdf********
16:19:32.915 -> getPacketData path_B Len = 2
16:19:32.915 -> getPacketData Path: R
16:19:32.949 -> getPacketData topic_B Len = 7
16:19:32.949 -> hit2
16:19:32.949 -> getPacketData len0 Topic: status
16:19:32.949 -> hit3
16:19:32.949 -> iot-2/evt/status/fmt/json{
16:19:32.949 ->   "DeviceID": "R",
16:19:32.949 ->   "MessageID": "y93xur9l",
16:19:32.949 ->   "Payload": "t**s*sdf********",
16:19:32.949 ->   "path": "R,Papa"
16:19:32.949 -> }
16:19:32.949 -> Publish to Cloud ok
16:19:32.949 -> {
16:19:32.949 ->   "DeviceID": "R",
16:19:32.949 ->   "MessageID": "y93xur9l",
16:19:32.949 ->   "Payload": "t**s*sdf********",
16:19:32.949 ->   "path": "R,Papa"
16:19:32.949 -> }
16:19:32.949 -> Publish To Local MQTT ok

Example of a more fully filled out form failing publish
16:52:04.810 -> handlePacket pSize 95
16:52:04.810 -> LORA RCV millis:2046266 rssi:-50.00 snr:6.25 fe:11785.99 size:95 data:f5025200f6096c3236627764306900f7415472656e742a42726f776e2a37333120526f626572746f204176652a53616e746120426172626172612a39333130392a383035373239313831312a2a2a2a2a2a00f3025200e30773746174757300
16:52:04.842 -> getPacketData senderId_B Len = 2
16:52:04.842 -> getPacketData User ID: R
16:52:04.842 -> getPacketData messageId_B Len = 9
16:52:04.842 -> getPacketData Message ID: l26bwd0i
16:52:04.842 -> getPacketData payload_B Len = 65
16:52:04.842 -> getPacketData Message: Trent*Brown*731 Roberto Ave*Santa Barbara*93109*8057291811******
16:52:04.887 -> getPacketData path_B Len = 2
16:52:04.887 -> getPacketData Path: R
16:52:04.887 -> getPacketData topic_B Len = 7
16:52:04.887 -> hit2
16:52:04.887 -> getPacketData len0 Topic: status
16:52:04.887 -> hit3
16:52:04.887 -> iot-2/evt/status/fmt/jsonPublish to 'client' failed
16:52:04.887 -> Publish to 'LocalMqttServer' failed

**To Reproduce**
Steps to reproduce the behavior:
Fill out the captive portal form on the MaMaDuck example and hit send. Monitor the serial output on the PaPaDuck.

**Expected behavior**
the Form information should be published regardless of the amount of information entered on the form

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

trentbrown13, 2020-08-06:
looks like 20 char max on the form.

20 chars pass
21 chars fail

20 chars entered on two fields pass
8:19:39.576 -> LORA RCV millis:7300955 rssi:-48.00 snr:6.00 fe:11685.33 size:62 data:f5025200f609766f3336316c773200f720303132333435363738392a303132333435363738392a2a2a2a2a2a2a2a2a2a00f3025200e30773746174757300
18:19:39.615 -> getPacketData senderId_B Len = 2
18:19:39.615 -> getPacketData User ID: R
18:19:39.615 -> getPacketData messageId_B Len = 9
18:19:39.615 -> getPacketData Message ID: vo361lw2
18:19:39.615 -> getPacketData payload_B Len = 32
18:19:39.615 -> getPacketData Message: 0123456789*0123456789**********
18:19:39.615 -> getPacketData path_B Len = 2
18:19:39.615 -> getPacketData Path: R
18:19:39.615 -> getPacketData topic_B Len = 7
18:19:39.615 -> hit2
18:19:39.615 -> getPacketData len0 Topic: status
18:19:39.615 -> hit3
18:19:39.615 -> iot-2/evt/status/fmt/jsonPublish to 'client' failed
18:19:39.649 -> Publish to 'LocalMqttServer' failed
18:20:36.170 -> handlePacket pSize 98
18:20:36.170 -> LORA RCV millis:7357544 rssi:-137.50 snr:-11.50 fe:21634.22 size:98 data:aaff7de396a2f3141e061486a973772aa31f18ecbaea68c258a1c682e434e81e8bb51eeaaa88998fe9c95995220b20ec0605571b17d5f2dde011862013280dea2cff5cacac78f998f4d3111409ad24fa8774c198dea97ed7085e772dcf02b54a30fb
18:20:36.205 -> getPacketData topic_B Len = 150
18:20:36.205 -> hit2
18:20:36.205 -> getPacketData path_B Len = 20
18:20:36.205 -> getPacketData Path: ⸮⸮⸮sw*⸮⸮⸮h⸮X⸮⸮
18:20:36.205 -> getPacketData Path: 
18:20:36.205 -> hit1
18:20:36.205 -> LORA SND millis:7357572 size:4 data:f8023100
18:20:36.239 -> startTransmit Packet sent
18:20:36.239 -> getPacketData pong sent
18:20:36.239 -> iot-2/evt//fmt/json{
18:20:36.239 ->   "DeviceID": "",
18:20:36.239 ->   "MessageID": "",
18:20:36.239 ->   "Payload": "",
18:20:36.239 ->   "path": "⸮⸮⸮sw*⸮⸮⸮h⸮X⸮⸮,Papa"
18:20:36.239 -> }
18:20:36.239 -> Publish to Cloud ok
18:20:36.239 -> {
18:20:36.239 ->   "DeviceID": "",
18:20:36.239 ->   "MessageID": "",
18:20:36.274 ->   "Payload": "",
18:20:36.274 ->   "path": "⸮⸮⸮sw*⸮⸮⸮h⸮X⸮⸮,Papa"
18:20:36.274 -> }
18:20:36.274 -> Publish To Local MQTT ok

21 chars entered on 2 fields failed
18:21:22.005 -> LORA RCV millis:7403371 rssi:-47.00 snr:6.50 fe:11685.33 size:63 data:f5025200f6096d32686d796d706700f721303132333435363738392a303132333435363738392a312a2a2a2a2a2a2a2a2a00f3025200e30773746174757300
18:21:22.005 -> getPacketData senderId_B Len = 2
18:21:22.005 -> getPacketData User ID: R
18:21:22.005 -> getPacketData messageId_B Len = 9
18:21:22.039 -> getPacketData Message ID: m2hmympg
18:21:22.039 -> getPacketData payload_B Len = 33
18:21:22.039 -> getPacketData Message: 0123456789*0123456789*1*********
18:21:22.039 -> getPacketData path_B Len = 2
18:21:22.039 -> getPacketData Path: R
18:21:22.039 -> getPacketData topic_B Len = 7
18:21:22.039 -> hit2
18:21:22.039 -> getPacketData len0 Topic: status
18:21:22.039 -> hit3
18:21:22.039 -> iot-2/evt/status/fmt/jsonPublish to 'client' failed
18:21:22.039 -> Publish to 'LocalMqttServer' failed
18:23:14.764 -> handlePacket pSize 116
18:23:14.764 -> LORA RCV millis:7516134 rssi:-138.00 snr:-12.00 fe:-27808.24 size:116 data:b2e4eb461af719991dcc3656c124c08b01887834990ecaaa7fbb0694e08220a8cf621f458cc0f013752d3c441814892160870fc393cbd18692962ade36284f0eff7da6141adb9bec9629170d9e3ccefed42adb9bbe119da0821a7d593db28f3d3c31df683755edb0688bea0da24c715d45374e85
18:23:14.798 -> getPacketData payload_B Len = 25
18:23:14.798 -> getPacketData Message: ⸮⸮6V⸮$⸮⸮⸮x4⸮ʪ⸮⸮⸮⸮ ⸮
18:23:14.798 -> iot-2/evt/status/fmt/json{
18:23:14.798 ->   "DeviceID": "",
18:23:14.798 ->   "MessageID": "",
[... 41 lines cut ...]
18:23:14.798 ->   "MessageID": "",
18:23:14.798 ->   "Payload": "⸮⸮6V⸮$⸮⸮⸮x4⸮ʪ⸮⸮⸮⸮ ⸮",
18:23:14.833 ->   "path": ",Papa"
18:23:14.833 -> }
18:23:14.833 -> Publish To Local MQTT ok
18:24:23.005 -> dhcps: send_offer>>udp_sendto result 0
18:24:23.005 -> dhcps: send_offer>>udp_sendto result 0
18:24:23.039 -> dhcps: send_nak>>udp_sendto result 0
18:24:23.039 -> dhcps: send_nak>>udp_sendto result 0
18:25:21.302 -> handlePacket pSize 63
18:25:21.302 -> LORA RCV millis:7642676 rssi:-47.00 snr:6.25 fe:11702.11 size:63 data:f5025200f60967366e366d6c377a00f7213031323334353637383930313233343536373839312a2a2a2a2a2a2a2a2a2a2a00f3025200e30773746174757300
18:25:21.336 -> getPacketData senderId_B Len = 2
18:25:21.336 -> getPacketData User ID: R
18:25:21.336 -> getPacketData messageId_B Len = 9
18:25:21.336 -> getPacketData Message ID: g6n6ml7z
18:25:21.336 -> getPacketData payload_B Len = 33
18:25:21.336 -> getPacketData Message: 012345678901234567891***********
18:25:21.336 -> getPacketData path_B Len = 2
18:25:21.336 -> getPacketData Path: R
18:25:21.336 -> getPacketData topic_B Len = 7
18:25:21.336 -> hit2
18:25:21.336 -> getPacketData len0 Topic: status
18:25:21.371 -> hit3
18:25:21.371 -> iot-2/evt/status/fmt/jsonPublish to 'client' failed
18:25:21.371 -> Publish to 'LocalMqttServer' failed

trentbrown13, 2020-08-06:
Ok, apparently I am running an older version of PubSubClient where the MQTT_MAX_MESSAGE_SIZE is set to 128 bytes rather than the newer default of 256 bytes.

Perhaps noting in the docs? It was a bit frustrating for a while and could be for others as well. You can note this ticket: https://github.com/knolleary/pubsubclient/issues/66

and this doc:
https://pubsubclient.knolleary.net/api.html#configoptions

trentbrown13, 2020-08-06:
perhaps could be included in the wiki>

closed 2020-08-06 by trentbrown13

krook, 2020-08-06:
Thanks for the field notes @trentbrown13. @tqrahman can we update the docs?

tqrahman, 2020-08-06:
Awesome, I added it to the FAQs in the wiki. Thank you @trentbrown13 and @krook!