ClusterDuck-Protocol/ClusterDuck-Protocol #117: Need to limit amount of chars allowed to be entered in the captive portal

trentbrown13, 2020-08-06:
**Describe the bug**
There is a limit to the allowed message size when sending. If it is exceeded the msg will not be sent and the Duck will PanicDuck. Unfortunately  the user will likely not realize their disaster msg did not get sent.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to the captive portal (I used the MamaDuck example config)
2. Enter a lot of char's in each field, especially the comments section. I just added a bunch of '1's in each field
 Hit send
9:22:37.776 -> Topic: status
09:22:37.776 -> uuidCreator 9xba1mpe
09:22:37.776 -> Warning: message is too large!
09:22:37.776 -> sendPayloadStandard packetIndex 1423
09:22:37.776 -> LORA SND millis:27293 size:1423 data:f5025200f60939786261316d706500f7713131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a31313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a3131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a31313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a312e3131313131313131313131313131313131652b39332a6f6b2a32342a31372a32332a31313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131313131312a00f3025200e30773746174757300
09:22:38.015 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
09:22:38.015 -> Core 1 register dump:
09:22:38.050 -> PC      : 0x400d67c7  PS      : 0x00060c30  A0      : 0x800d47d6  A1      : 0x3ffd8ca0  
09:22:38.050 -> A2      : 0x00e67ec0  A3      : 0x00e67ec0  A4      : 0x00e67ec0  A5      : 0x00000000  
09:22:38.050 -> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8008c0ba  A9      : 0x3ffd8cc0  
09:22:38.050 -> A10     : 0x3ffc8d78  A11     : 0x3ffdfff9  A12     : 0x3ffdf0f4  A13     : 0x3ffdfc24  
09:22:38.050 -> A14     : 0x3ffdeb55  A15     : 0x3ffdfc25  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
09:22:38.083 -> EXCVADDR: 0x00e67ec4  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
09:22:38.083 -> 
09:22:38.083 -> Backtrace: 0x400d67c7:0x3ffd8ca0 0x400d47d3:0x3ffd8cd0 0x400d4871:0x3ffd8cf0 0x400d4937:0x3ffd8d20 0x400d57ff:0x3ffd8d40 0x400d615f:0x3ffd8d60 0x400d342a:0x3ffd8da0 0x400d36ec:0x3ffd8dd0 0x400d3aa6:0x3ffd8e50 0x400ddd05:0x3ffd8ef0 0x400dc0d1:0x3ffd8f30 0x400dc161:0x3ffd8f70 0x400deef9:0x3ffd8f90 0x400def79:0x3ffd8fd0 0x400df58a:0x3ffd8ff0 0x40088b7d:0x3ffd9020
09:22:38.117 -> 
09:22:38.117 -> Rebooting...

**Expected behavior**
Each field should have a reasonable limit on the number of allowed char's and the sum of the field lengths should be <= total allowed number.

krook, 2020-08-10:
Thoughts on this @nfeuer @TimoWielink? Seems pretty high priority.

knouse1344, 2020-08-10:
@Evilgeniusnerd implemented some level of a solution in the Ducks in his house...Zach any thoughts?

nfeuer, 2020-08-17:
We need to enforce a 230 character limit on the captive portal

TimoWielink, 2020-08-31:
urrently working on Character counter in the html file that allows a max count of characters. Need to merge new PR for PubSub version 2.7 that fixes 256 character limit for MQTT so we have no errors there.

knouse1344, 2020-09-08:
it would be great to include "captive portal options" as like an a-la-carte selection.  think like menu items do you want "this one?" or "this one" ? or "this one" ? with different layouts, buttons, some text, some checkboxes, etc.... then could help folks who don't want to code an entire new one from scratch.

in meantime, timo and I solving the payload length issue in js.

knouse1344, 2020-09-11:
so it seems like it would be useful to do a couple things.

1. keep the ability to create any captive portal you wish
2. Either prevent, or automatically detect and prepare a size of data that expands beyond 1 packet size.  Either need to have developers understand this when building HTML, or prevent it from happening in the user experience.   I feel like ultimately, multi-packet deliveries make the most sense (and provide most flexible future for development) but i can see points to either side.

nfeuer, 2020-09-11:
We should open a new issue addressing flexible captive portal options. @dmandala wants to have unlimited characters and chop the message into multiple packets. Timo's most recent commit fulfills this issue

closed 2020-09-11 by nfeuer