ClusterDuck-Protocol/ClusterDuck-Protocol #123: A clean install on Ubuntu 20.04 (.01) fails to build

dmandala, 2020-08-29:
When following the document: https://github.com/Project-Owl/ClusterDuck-Protocol/blob/master/doc/howToBuildADuck.md (and correctly bypassing the bug in the manual install section) you start the IDE, select the proper board (in my case the Heltec LoRa 32 v2), load the example code "MamaDuckExample" and compile.  

The compile will fail with the error: **"exec: "python": executable file not found in $PATH"**

This is because the python module pyserial is not installed.  Worse, you must install the python2 version not python3.  This is worse because python2 is no longer supported and is not installed by default on Ubuntu and it is fully DEPRECATED and all support will be removed January 2021 and will become unavailable at all.

So to enable compilation you have to:
1) Install python2: `sudo apt install python`
2) Install pip2 step 1: `sudo apt install curl`
3) Install pip2 step 2: `curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py`
4) Install pip2 step 3: `sudo python2 get-pip.py`

> DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support

5) Install pyserial: `pip2 install pyserial`

> DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support

Now the build will work.  BUT notice the above "DEPRECATION" warnings. January 2021 this fix will stop working at all.

Short term fix is to document the above steps.  Long term fix is to remove all dependencies on Python2 within the build code base.  This does not happen when you build for a standard Arduino Uno (or any Arduino hardware) so the issue is brought in with the hardware or the libraries I don't know which.

Now once you fix the python2 vs python3 the instructions for loading pyserial are much easier:
1. Install python pip3: `sudo apt install python3-pip`
2) Install pyserial module: `pip3 install pyserial`

 Lots less steps. But will only help once the dependency on python2 is fixed.

I'll describe how I did the install:
To get an absolutely clean starting point I created a VirtualBox VM using Ubuntu 20.04.01.  I did this so I can resolve some other bugs caused by library name collisions.
1) Create a 64but Linux VM pointing at the ISO for Ubuntu 20.04.01 freshly downloaded. Choose the defaults except use the small install choice.  No need for word processors, etc.
2) Using firefox open the document at: https://github.com/Project-Owl/ClusterDuck-Protocol/blob/master/doc/howToBuildADuck.md
3) Following the instructions download a 64 bit Linux tarball from https://www.arduino.cc/en/main/software and untar into /opt  Use the two install scripts (arduino-linux-setup.sh & install.sh) to full install the IDE. _No instructions really provided for this beyond the URL provided._
4) Continuing to follow the instructions start the IDE and open the preferences and paste https://dl.espressif.com/dl/package_esp32_index.json,https://adafruit.github.io/arduino-board-index/package_adafruit_index.json into the Additional Boards Manger URLS:
5) Continuing to follow the instructions to the section: "Install ClusterDuck Protocol" "Manual Install"
6) Continuing to follow the instructions to the section: "Load the Heltec ESP32 Board to your Arduino IDE:" searching for esp32 and installing the library.
7) In the IDE select" `File->Examples->ClusterDuck->MamaDuckExample`
8) Compile and fail with the above noted error: **"exec: "python": executable file not found in $PATH"**.

Additionally it was noted that even in this brand new install there was a library name collision: 

> Multiple libraries were found for "WiFi.h"
>  Used: /home/davidm/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
>  Not used: /opt/arduino-1.8.12/libraries/WiFi

This really should be addressed, even in a totally clean install we have collisions; do we really need this extra WiFi library?  If so can we rename it working with the author?

Evilgeniusnerd, 2020-08-29:
@dmandala great information here will get working on some of these things. As for the duplicate WiFi modules, it has been an ongoing issue for a while. We can just get rid of it we don't need both of them.

Evilgeniusnerd, 2020-09-05:
We have the script and documentation https://github.com/Code-and-Response/ClusterDuck-Protocol/wiki/Linux-CDP-Dev-Setup  That now solves all these issues mentioned above. Please feel to add specific Linux distro support or additional features to this.

closed 2020-09-05 by Evilgeniusnerd