ClusterDuck-Protocol/ClusterDuck-Protocol #156: Header.html not recognized in Doxygen (Github Actions Fails)

TimoWielink, 2020-12-09:
On my local machine it generates the Doxygen with no compile errors. The GitHub action fails since it can't find the header.html file which is linked here: 
[Doxygen Header.html import](https://github.com/Call-for-Code/ClusterDuck-Protocol/blob/cdc517ed900a9bfc5b9bfb0f56da570d3c4ba5e8/docs/Doxyfile#L214)

One line underneath imports the footer.html which seems to be working fine. 

I need to look into if there is a difference between header and footer or if we need to stick with the normal header.

[Header.html](https://github.com/Call-for-Code/ClusterDuck-Protocol/blob/master/docs/header.html)

Evilgeniusnerd, 2021-04-17:
@TimoWielink what's the status on this?

TimoWielink, 2021-04-20:
We can resolve this since i took it out

closed 2021-04-20 by TimoWielink