ClusterDuck-Protocol/ClusterDuck-Protocol #167: Papa Duck will not receive nor transmit messages from Mama Ducks unless there is a successful MQTT connection established

trentbrown13, 2021-01-29:
Symptom:
	Papa Duck does not receive, display or forward messages from Downstream Ducks. This is evidenced by no serial monitor activity at all when a Mama Duck sends a message. The Papa Duck will still continue to display its own internal messages such as "udp_sendto" messages.
	
	Unfortunately there are no serial monitor notifications if the mqtt client connection fails (it fails silently) , though there is a notification if it succeeds.
	
Debugging:
	1: Verify Lora connection by setting two Mama ducks and verifying the message is transmitted (great suggestion from Nick)
	2: In PapaDuck.cpp -> void setup_mqtt(bool use_auth) add an else statement to print an error if the mqtt client has not connected to the server. This will let you know if the most likely failure reason is that the client has not connected to the mqtt server (usually an error in wifiClient and pubsubClient arguments and/or the server config).
	
Resolution:
	1: Follow the wiki and either setup DMS light or the DMS cloud version that will allow mqtt to connect with the default example.
	2: Setup your own mqtt broker. This can be done very easily following many online tutorials, most often using an RPI.
		In PaPaDuck
			// Set the mqtt broker address to your local broker rather than the default in the example
			//char server[] = ORG ".messaging.internetofthings.ibmcloud.com";
			char server[] = "your mqtt broker address";
			
			// If you don't use auth mode on your local server unset it    
			
			//bool use_auth_method = true;
			bool use_auth_method = false;
			
			// Change WiFiClientSecure to WifiClient

			//WiFiClientSecure wifiClient;
			WiFiClient(wifiClient);
			
			// Change PubSubClient to reflect new values and udp port (mqtt default = 1883)
			PubSubClient client(server, 1883, wifiClient);
			
Notes:
	The mqtt connection should probably be non-blocking and additional error messages added if the connection fails (as part of setup). The documentation does not mention that an MQTT connection is required for the Papa Duck to function correctly. Documentation for using your own MQTT broker can be added.

TimoWielink, 2021-01-29:
Thankk you, wokring on this!

nfeuer, 2021-02-15:
@trentbrown13 this was actually fixed recently with the patch fix here #170

closed 2021-02-15 by nfeuer