ClusterDuck-Protocol/ClusterDuck-Protocol #179: CDP fails to build for Arduino Zero (USB) based boards

amirna2, 2021-03-07:
**Describe the bug**
When attempting to build the latest version of CDP from the main branch I ran into an issue where
DuckDisplay.h build is failing due to the compiler not able to resolve `std::vector`

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure you have the latest version `commit: 2b03c5bd7f6ffa86cc2427610ba91d66396dcb3e`
2. Select Arduino Zero (USB) as you board from the Arduino IDE
3. Open the DuckLink.ino example and click Verify (this will compile the code, without installing it)
4. See error
```
Arduino: 1.8.13 (Mac OS X), Board: "Arduino Zero (Native USB Port)"
In file included from /Users/anathoo/Documents/Arduino/libraries/ClusterDuck-Protocol/src/include/DuckRadio.h:14:0,
                 from /Users/anathoo/Documents/Arduino/libraries/ClusterDuck-Protocol/src/include/DuckNet.h:22,
                 from /Users/anathoo/Documents/Arduino/libraries/ClusterDuck-Protocol/src/include/Duck.h:5,
                 from /Users/anathoo/Documents/Arduino/libraries/ClusterDuck-Protocol/src/DuckLink.h:7,
                 from /var/folders/57/0h65vzcn5px_061yp_bmc31r0000gn/T/arduino_modified_sketch_841209/DuckLink.ino:9:
/Users/anathoo/Documents/Arduino/libraries/ClusterDuck-Protocol/src/include/../DuckDisplay.h:40:40: error: 'std::vector' has not been declared
   void setupDisplay(int duckType, std::vector<byte> name) {}
                                        ^~~~~~
/Users/anathoo/Documents/Arduino/libraries/ClusterDuck-Protocol/src/include/../DuckDisplay.h:40:46: error: expected ',' or '...' before '<' token
   void setupDisplay(int duckType, std::vector<byte> name) {}
                                              ^
exit status 1
Error compiling for board Arduino Zero (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

**Expected behavior**
The DuckLink example should build without error for the Arduino Zero compatible board as it is one of the CDP supported hardware.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Mac OSX
 - Arduino IDE: 1.8.13
 
**Additional context**

Suggest fix is to explicitly include the vector header as some Arduino libraries fail to resolve `std::vector`.
With the change below, I was able to compile the DuckLink example for the RocketScream Arduino M0 compatible board.

```
diff --git a/src/DuckDisplay.h b/src/DuckDisplay.h
index f110cc3..602e89e 100644
--- a/src/DuckDisplay.h
+++ b/src/DuckDisplay.h
@@ -17,6 +17,7 @@
 #include <WString.h>
 #include "include/DuckTypes.h"
 #include "include/DuckEsp.h"
+#include <vector>
 #ifndef CDPCFG_OLED_NONE
 #include <U8g2lib.h>
 #endif
 ```

nfeuer, 2021-03-07:
Will try that out. We will also add the other supported boards to the Travis build so we can catch this sooner. Thanks for the heads up

amirna2, 2021-03-08:
@nfeuer The CI pipeline is a great addition.
I have a local branch with the fix: https://github.com/amirna2/ClusterDuck-Protocol/tree/fix/missing-std-vector-header-arduino-zero

Evilgeniusnerd, 2021-04-17:
@nfeuer have these changes been added to patch this issue?

closed 2021-06-21 by nfeuer