ClusterDuck-Protocol/ClusterDuck-Protocol #197: Wiki - fix installation instructions

doniks, 2021-05-22:
To which page do you want to add to the wiki?
- edit linux install page

What do you want to change/add?
Currently it says:

sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10>

The > at the end gives an error. Pretty sure this is just a typo. I suggest to remove the >

Is your documentation related to a recent pull request or issue?
no

doniks, 2021-05-22:
here https://github.com/Call-for-Code/ClusterDuck-Protocol/wiki/Linux-CDP-Dev-Setup#preferred-method-of-setup-script-install

Evilgeniusnerd, 2021-05-24:
@doniks  indeed was a typo in the command. Now works, and I updated the wiki page. Let me know if you run into any other issues while following that page.

closed 2021-05-24 by Evilgeniusnerd

nfeuer, 2021-05-24:
Thank you @doniks for pointing that out!