ClusterDuck-Protocol/ClusterDuck-Protocol #209: deviceID in the payload is corrupted but not corrupted in the path

tqrahman, 2021-06-08:
You can see that the deviceIDs (first column) are corrupted. However if you look at the path (last column), the first 8 characters in path is the deviceID of that message and it is NOT corrupted.

[image]

amirna2, 2021-06-23:
Not sure if this is happening still, but it could either be an issue in PapaDuck.ino or on the receiving end of the JSON payload in the cloud.

nfeuer, 2021-10-04:
This still happens, but much less with the recent updates. Might be a server-side issue. Should do more testing

nfeuer, 2022-03-10:
Should we add another CRC or change how the existing one works? Its very strange that these kinds of things get through

faradaym, 2024-08-26:
This issue happened in CDP v2, not observed so far in v4

closed 2024-08-26 by faradaym