ClusterDuck-Protocol/ClusterDuck-Protocol #216: Compilation errors for Heltec ESP32 WiFi v2 board

amirna2, 2021-06-20:
**Describe the bug**
When building CDP for Heltec ESP32 v2 board I got a couple of compilation errors stating methods without return statements

```
/home/amir/Arduino/libraries/ClusterDuck-Protocol/src/DuckNet.cpp: In member function 'int DuckNet::saveWifiCredentials(String, String)':
/home/amir/Arduino/libraries/ClusterDuck-Protocol/src/DuckNet.cpp:285:3: error: no return statement in function returning non-void [-Werror=return-type]
   }
   ^
/home/amir/Arduino/libraries/ClusterDuck-Protocol/src/DuckNet.cpp: In member function 'int DuckNet::loadWiFiCredentials()':
/home/amir/Arduino/libraries/ClusterDuck-Protocol/src/DuckNet.cpp:319:1: error: control reaches end of non-void function [-Werror=return-type]
 }
```

**To reproduce**
Steps to reproduce the behavior:
1. checkout CDP from master branch (v2.9.12)
2. select Heltec ESP32 WiFi v2 as the board
3. compile CDP
4. the compiler throws error above

**Expected behavior**
Compilation should be successful

TimoWielink, 2021-07-18:
@amirna2 Can i close this?

amirna2, 2021-07-31:
Yes

closed 2021-10-04 by nfeuer