ClusterDuck-Protocol/ClusterDuck-Protocol #231: Move EEPROM methods out of DuckNet

TimoWielink, 2021-07-22:
**Describe the bug**
All the EEPROM methods are built inside of DuckNet since that's where originally passed into the value for the EEPROM. 

**To reproduce**
Open Ducknet.cpp and look at:

`saveWifiCredentials()`
`loadWifiCredentials()`
`saveControlSsis()`
`saveControlPassword`
`loadControlSsis()`
`loadControlPassword`

Move these methods to DuckUtils

**Additional context**
More EEPROM methods are being built now for saving channels, which reminded me of moving them.

l-monninger, 2021-09-24:
I'm new to the project. These seems like a pretty mellow first contribution. Is this still needed?

nfeuer, 2021-10-01:
Hi @l-monninger, sorry for the delay! Yes this is still a need!

nfeuer, 2021-10-11:
Sure! You can join our slack if you need any help www.project-owl.com/slack

closed 2021-10-30 by nfeuer