ClusterDuck-Protocol/ClusterDuck-Protocol #234: Clean up startReceive

rolsen, 2021-07-29:
Duck has startReceive and DuckRadio has startReceive. Duck's version calls duckesp::restartDuck() if there's an error from lora.startReceive(), but DuckRadio's version doesn't.

It makes the most sense to me to have DuckRadio be the only object that can startReceive, and to have it take any restart/mitigation steps as needed.

I imagine startReceive is unlikely to return an error assuming there's no synchronization issues (where startReceive is called when `transmit()` is in progress, for example, though I didn't test that).

nfeuer, 2021-10-04:
This should be moved to DuckRadio

amirna2, 2024-03-06:
This has been fixed for quite some time now.

closed 2024-03-06 by amirna2