ClusterDuck-Protocol/ClusterDuck-Protocol #240: Unique IDs for Duck hardware.

BrentonPoke, 2021-09-27:
The software currently requires the developer to manage IDs for individual Ducks by naming them manually, but something important we need to do is find a way of programmatically provisioning unique names in order for the routing algorithm to function properly. There are specific constraints that need to be observed:

- 8 bytes in length
- deterministic generation
- unique under any hardware condition
- implemented as a utility function within core ClusterDuck library

The deterministic requirement is important because it maintains fault tolerance in the event that a Duck goes offline and restarts again. A probabilistic generator will mean a new ID every time this happens, causing problems for routing. The 8 bytes is also non-negotiable because the packet has 48 bytes reserved for IDs, allowing for 6 routes. This It is tempting to use hardware IDs like MAC addresses, but those are sometimes set with the same NIC identifier for all devices from the same manufacturer. This is not to say the MAC address can't be used as a seed to a function in some way. However this is implemented, it can't be a function the developer is responsible for managing. Provisioning a device with a deterministic unique ID needs to be handled out-of-the-box and opaque to the developer. So that means the person programming a duck should not have to remember to call the function, it should be a utility function that we will use under the hood.

l-monninger, 2021-09-28:
Are you thinking we hash hardware a la [this](https://github.com/Tarik02/machineid/blob/master/src/machineid/machineid.cpp)? What about using something like [BIOS UUID](https://www.codeproject.com/Tips/5263343/How-to-Get-the-BIOS-UUID)?

amirna2, 2021-09-28:
The Device unique ID is an interesting problem to solve. The reason behind the 8 bytes is that it allows us to follow the LoRaWAN Device EUI model if necessary. Also many LoRaWAN compatible hardware (e.g Radio chips) will already have a device ID we can use.

See the specification here: https://lora-alliance.org/wp-content/uploads/2020/11/TR005_LoRaWAN_Device_Identification_QR_Codes.pdf

Not saying we should implement that part of LoRaWAN, but the intent behind the Device UID is to
1. allow pairing of a device and its safe provisioning when it joins a cluster for a specific application.
2. be used for encryption.
For these reasons, typically the Device ID is flashed in secure memory at factory time. The device manufacturer maintains a Database of all devices produced. When a device is purchased, it can be verified against the database when it is activated.

The problem with having the Duck generate its own device UID is that we can't guarantee it is a known device.

l-monninger, 2021-09-28:
@amirna2 Why do we need to guarantee that it is a known device?

amirna2, 2021-09-28:
Because one could impersonate a duck and join your cluster to hack into your network or cause harm.
If you don't have a way to know the device joining the network is one that is allowed to join, it's taking a significant security risk.

l-monninger, 2021-09-28:
Sorry, not the most precise way to ask that question and very new to project. Why do we need the ID @BrentonPoke  is suggesting to guarantee that it is know device? Is the check for a known device the same handled by reviewing the ID currently? If so, aren't we already (and perhaps more so) vulnerable to impersonation?

amirna2, 2021-09-28:
CDP currently doesn't have any mechanisms to perform device validation or for a device to securely be provisioned.
What Brenton is referring to is that currently we simply hard code some 8 bytes value (in form of a string). Now that's fine for playing around and as long as that device doesn't leave your lab you are okay. What needs to happen is we no longer hardcode the ID and chose one "by hand".
One way of doing this (and how it would be done in a product) is the manufacturer or owner of the device would provide that ID at factory time. These ID would be allocated by an authority like IEEE for example.
The key here is the ID should be already present and accessible to the developer who is going to work on the Duck firmware, not the developer coming up with some kind of algorithm to define what that ID should be, otherwise one could reverse engineer the firmware and find how to craft device IDs. And obviously hard coding one makes the task even simpler.
Typically a Device ID would have some parts used as a "meta data" that can be decoded, like Manufacturer Identification, device class, etc... and some parts would be randomly generated in advance as part of a pool of numbers, then associated to a device in factory.

amirna2, 2021-09-28:
Here is an other document that outlines security features for LoRa devices:
https://lora-alliance.org/wp-content/uploads/2020/11/la_faq_security_0220_v1.2_0.pdf
Specifically point #5 is of interest.

l-monninger, 2021-09-28:
Thanks! @amirna2 

I guess this is where my confusion is because I believe this is a fundamental problem in networking that is only really resolved by an authority, as you mentioned. And, even then, the _identities_ are spoofable which is why they instead are addresses.

Now networking is not my area of expertise (if there is one); here to broaden my horizons in part. But, to me, it seems like the concerns you are raising lead to a line of questions about how to implement zero trust features. As far as I know, those are handled by gateways and Layer 7 technology. And, I wouldn't know if it's reasonable to include those in CDP.

l-monninger, 2021-09-28:
Also, do you think this is worthy of `good first issue`?

amirna2, 2021-09-28:
My initial idea was that the device ID if kept secret, could participate in the encryption schema.
There is actually a different approach to this where Device UID is not necessarily kept secret. 
The Device UID could also be put on a label, QR code, etc... to help with simple identification during the provisioning of the device. It essentially would be similar to how we use mac addresses.
When the device is first powered, it advertise itself on the network and request to join a cluster, as part of that the Device ID is provided. The key exchange for establishing end to end encryption is a completely separate topic.

Again referencing the LoRaWAN spec and join procedure for some inspiration and knowledge sharing:
https://www.centennialsoftwaresolutions.com/post/deveui-appeui-joineui-and-appkey.

In the light of this, I am not totally sure what we need at this point. As you said, generation of the Device ID isn't really part of the protocol. It's something that must be provided to the device and stored in it (and possibly shared for ease of device management and provisioning).

l-monninger, 2021-09-28:
It is fun food for thought though. Have you ever heard of [RAIN](https://books.google.com/books?id=CWrnDwAAQBAJ&pg=PA66&lpg=PA66&dq=Routing+algorithm+for+network+of+homogeneous+and+id-less+biomedical+sensor+nodes+(RAIN)&source=bl&ots=sHmJqALMoD&sig=ACfU3U1cZA02ki-bYafrSAka1gtTEPMeGw&hl=en&sa=X&ved=2ahUKEwiTh82-1qHzAhUjL30KHcpLDqkQ6AF6BAgPEAM#v=onepage&q=Routing%20algorithm%20for%20network%20of%20homogeneous%20and%20id-less%20biomedical%20sensor%20nodes%20(RAIN)&f=false)? I guess it's used in id-less BAN. Wondering if there are any learnings we could apply here, if we were interested.

BrentonPoke, 2021-09-28:
> Also, do you think this is worthy of `good first issue`?

It is just one function and could be done any undergrad with their C++ courses completed. It would also be great experience for that student.

l-monninger, 2021-09-28:
@BrentonPoke 

> > Also, do you think this is worthy of `good first issue`?
> 
> It is just one function and could be done any undergrad with their C++ courses completed. It would also be great experience for that student.

Per above, it doesn't seem the function to get a unique machine ID is really the issue though. It seems there's a big question mark on how the ID would actually be used.

BrentonPoke, 2021-09-29:
There isn't a big question mark, Amir is talking about something long term that has to do with LoRaWAN. The ID is to be used to identify a route node in the packet structure.

l-monninger, 2021-09-29:
Okay, just want to make sure I have all my ducks in a row. Do you assign issues here or is it just whoever submits a working PR first?

krook, 2021-09-29:
Thoughts on how to go forward @nfeuer @TimoWielink?

BrentonPoke, 2021-09-30:
One thing I can put in the issue is the list of relevant files that will be of interest to an implementer. Will need to be brought up to speed on recent changes (which need to be documented), though.

l-monninger, 2021-10-08:
Where are we on this?

nfeuer, 2021-10-11:
One way we could do this is by taking a MAC address and putting it through some kind of hash function

krook, 2021-10-11:
@nfeuer did you already look into UUID generation? Or was this also an issue for space given that it's 32 bytes? http://www.ardulink.org/how-to-set-unique-id-to-arduino/

amirna2, 2024-03-03:
This is an old discussion, but still relevant. I want to summarize here what this DeviceID should be used for and why it is crucial to CDP.

The DeviceID is a unique identifier for the Duck. And it should be something that is internal to the Duck and only known to the Duck and the organization managing/owning that Duck. It is how primarily, a duck would be allowed to join a network.

Currently we hardcode a DeviceID in the Arduino sketch, this is okay for testing and experimenting. 

But in a real world application, the DeviceID should be something generated during manufacturing and stored in a secure flash location on the device. The device ID is then stored in a central database by the organization that owns the duck.

In that real world scenario, when the duck is first coming alive it sends the deviceID along with other information to the PapaDuck in the cluster. The Papa Duck relays that information to the organization Device Management System that can then lookup if the Duck is actually one that is registered in the DB. If that is the case, a message can be send back to the Duck to tell it to join the network.

All of this of course has to be exchanged through encrypted messages. 
Most Arduino device actually do have a `Chip ID` which can be used as a DeviceID:

```
const uint64_t chip_id = getID();
std::vector<byte> dev_id;
for (int i = 7; i >= 0; i--) {
    uint8_t val = (chip_id >> (i * 8)) & 0xFF;
    dev_id.push_back(val);
}
``` 

For ESP32 boards we can also use their MAC address
```
  uint32_t chipId=0;
  for(int i=0; i<17; i=i+8) {
    chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
  }
 ```

BrentonPoke, 2024-03-10:
`uint8_t val = (chip_id >> (i * 8)) & 0xFF;`
I'm not sure about this bitwise operation. What does the getID() function return that requires this?

amirna2, 2024-03-10:
The bitwise operation is pretty standard way to serialize a 64 bit value. I am just extracting each byte out of the chipID one by one to add it to the deviceID byte vector.

The getID() method is an Arduino specified method, but each core implements it differently based on the micro-controller capabilities. It does generally return a 64bit value, but often internally it is represented as 2x 32bits. That's the case on the Heltec CubeCell for example. 

But then on the ESP32 I just used the MAC address.