ClusterDuck-Protocol/ClusterDuck-Protocol #256: Add support for RadiLib 4.6.0

TimoWielink, 2021-11-16:
**Describe the bug**
In the new version of RadioLib, they fixed error handling for the SX1276, This means that our DuckRadio.cpp breaks when declaring our error messages.

**To reproduce**

1. Install RadioLib 4.6.0 
https://github.com/jgromes/RadioLib/releases/tag/4.6.0
2. Compile a Duck example
3. look at compilation error

**Screenshots**
[image]

**Additional context**
Working on a patch with @nfeuer , also need to prepare for Radiolib 5.0 release

r41d, 2022-05-16:
RadioLib 4.6.0 seems to be working fine as far as I can tell.
What doesn't work is RadioLib version 5 because it introduced breaking changes.
Shall we dedicate this issue to porting CDP to RadioLib 5 ?

TimoWielink, 2022-05-16:
Cool, Yeah It's been a while since I checked that. What LoRa chipset did you test on. I will look back into this

r41d, 2022-10-05:
This should be obsolete via #280

TimoWielink, 2022-10-05:
Thank you @r41d

closed 2022-10-05 by TimoWielink