ClusterDuck-Protocol/ClusterDuck-Protocol #258: Confuse about broadcastAck()

eydonmlau, 2021-11-26:
Hi,
    CDP is a great project, I'm a little confuse about the papaduck part which send broadcast ack back, Version 3.0.2
    In this function
  `void PapaDuck::storeForAck(const CdpPacket & packet) {
  ackStore.push_back(std::pair<Duid, Muid>(packet.sduid, packet.muid));
}`
  when papaduck received a packet, it store the packet.sduid and packet.muid, but no matter how many times this message frame is hopped，the packet.sduid will not changed, isn't it ? but in this function
`void PapaDuck::broadcastAck() {
  assert(ackStore.size() <= MAX_MUID_PER_ACK);
  const byte num = static_cast<byte>(ackStore.size());
  std::vector<byte> dataPayload;
  dataPayload.push_back(num);
  for (int i = 0; i < num; i++) {
    Duid duid = ackStore[i].first;
    Muid muid = ackStore[i].second;
    logdbg("Sending ack to DUID " + duckutils::toString(duid)
      + " for MUID " + duckutils::toString(muid));
    dataPayload.insert(dataPayload.end(), duid.begin(), duid.end());
    dataPayload.insert(dataPayload.end(), muid.begin(), muid.end());
  }`
Is it necessary to write all the received packet which has the same duid and muid？I'm little confuse about this.
And report the bug in CdpPacket.h
`#define MAX_MUID_PER_ACK 14`
Maximum number of DUID/MUID pairs is:
  floor( (maximum data payload size - N) / (DUID size + MUID size) )
Which is:
  floor( (180 - 1) / 12 ) = 14
I think maximum is no longer 180, but 229 while using bloom filter instead of  using path?

nfeuer, 2021-12-06:
Thanks! Will add this to the upcoming patch

nfeuer, 2022-03-10:
This is has been included in this patch

closed 2022-03-10 by nfeuer