ClusterDuck-Protocol/ClusterDuck-Protocol #274: Routing over Reticulum?

samuk, 2022-03-16:
**Describe the bug**
Transport level security vulnerabilities: https://www.youtube.com/watch?v=LBd-ewnjn8M

**To reproduce**
Steps to reproduce the behaviour:
See: https://www.youtube.com/watch?v=LBd-ewnjn8M

**Expected behaviour**
Messages are secure

**Additional context**

Where secure messaging is needed could you route the packets over reticulum: https://unsigned.io/projects/reticulum/

nfeuer, 2022-03-16:
So I have seen Reticulum before and have been meaning to take a deeper dive. Looked really interesting. Been reading Mark Qvist's blog every now and then. @samuk have you used Reticulum before?

samuk, 2022-03-16:
TBH I've been waiting for the Android & ESP32 stuff which has only come out in the last week or so. I plan to give it a go soon,

nfeuer, 2022-03-18:
Definitely share your thoughts when you try it out. I think I would like to bring this up as a topic during our next town hall

closed 2024-08-26 by faradaym