ClusterDuck-Protocol/ClusterDuck-Protocol #277: BMP280 Example

ja2wiles, 2022-04-23:
I'm trying to load the BMP280 example, but don't see what pins I should use for connect the sensor.
I'm using a Heltec ESP-32 WiFi LoRa board.  
I see on the PinOuts they use the following for I2C.
 - Pin 15 for OLED_SCL
 - Pin 4 for OLED_SDA.

What pins should I use for this example?
Thanks
Jeff

tqrahman, 2022-04-23:
Hey Jeff,

Try connecting:
  - Pin 21 -> SDA
  - Pin 22 -> SCL
Let me know if that works.

ja2wiles, 2022-04-26:
Thank you that worked.

I have been able to build the IBM Watson IoT Platform and have 1 Papa Duck and 2 different mama ducks (sensors) sending data into the IoT.

Do you have examples of expanding on this to build a more elaborate web interface using this collected data.   Would it be easier to use the DMS LITE process?

I currently have ESP32 devices collecting temperature and vacuum in maple syrup production.  I am connecting WiFi and sending a data to local (Pi) MQTT  Server.  Lora LAN should solve my connectivity and range issues.

Thanks
Jeff

nfeuer, 2022-04-28:
Hi @ja2wiles,

You could use DMS Lite to use as a front end. Would just have to host it online and hook up your own database to it. It has some python scripts that you can look at to help you with the DB schema that works with it. It uses SQLite so you might have to edit the queries slightly depending on what DB you want to use. 

Would be interested to learn more about your project too! Do you make your own maple syrup? 

Best,
Nick

hawk4031, 2022-04-28:
To piggyback on what @nfeuer said: we have some outstanding changes in our `dms-lite-docker` repository in a branch that will include a more robust database solution (MariaDB specifically). That repo also uses Docker containers with Docker Compose for setting up and running the solution, so should be pretty easily portable between clouds or even to run on your own hardware. 

I can help you with this if you're interested.

ja2wiles, 2022-04-29:
Thanks for the replies.   Yes we make maple syrup in Ohio.  This is done at a church camp as a fund raiser.  www.phoc.org  We have 700 trees and 900 taps all connecting with plastic tubing running running down a hillside.  The tubing is all under a vacuum, this increases the sap production.    I'm monitoring the temperature and  vacuum levels at the end of 7 main lines out in the woods.   

In the woods I have a battery powered wifi access point.  (issues with connectivity and distance)
At the end of lines I have built an Arduino with temperature and vacuum (wemos D1 mini) 
-battery powered, small solar panel and wifi enabled
Once a minute connects to wifi and sends a MQTT message to Raspberry Pi running mosquito. 
Raspberry Pi acts as webserver and I'm using simple java script in a webpage to subscribe to the mqtt server. 
- this should be written to a DB and saved.

Connectivity is my biggest issue.  I've tested lorawan and I can cover the whole woods and not need wifi access points. 
Running a local copy on a Raspberry Pi sounds the best to me.  I tried building the dms-lite on the Pi, but had some issues with the database.  I was able to get it running on a windows computer with docker.   

I would be interested in trying the dms-lite-docker with the updated DB when that is ready. 
My next task would be to build a web interface to display the temperatures and vacuum.  

I've include some pictures. 

[image]
[image]

Thanks for your interest
Jeff

hawk4031, 2022-04-29:
@ja2wiles I'd love to work with you on DMS Lite Docker. I made some small modifications to that repo last night and we have everything moved into the main branch there. As it all containerized, it should be a good solution for you. I have personally tested it on a Raspberry Pi, as well as x86-based machines with no problems in either case. The components used are quite similar to what you mentioned above: MQTT server with Mosquitto, a web front-end, a database for storage (we're using MariaDB in this case), and some Python scripts for moving data around between components. 

The web front-end currently doesn't do much more than just show the data as it arrives, but could be easily extended to include visualizations or other useful bits of information. If you are in the Slack community, feel free to ping me directly!

amirna2, 2022-05-21:
@ja2wiles this is a really cool project.
I created a very simple demo leveraging dms-lite that creates a backend in go to handle MQTT messages. It also creates an Sql-Lite db.
Feel free to use this if it's something that can help:

https://github.com/amirna2/cdp-demo

knouse1344, 2022-05-21:
@ja2wiles agreed with @amirna2 , this project is fascinating.  I have a couple questions:

- do you have any connectivity at a base station?  Cellular or Wifi?  
- DMS Lite is one solution, DMS cloud could be another (and possibly easier to use)
- Would it be useful to have pressure sensors in your tubes as well?  I suspect a BMP180 could be a cheap way to accomplish this.

Would love to help you develop this more if you are open to it.

ja2wiles, 2022-05-21:
Thanks for all of the input and help

Here is where I’m at.

  *   I have dms-lite running on a Pi
  *   I have built a sensor and it is sending data and I can see it in the dashboard web page.
  *   I’m using Adafruit MPRLS Ported Pressure Sensor
  *   https://www.adafruit.com/product/3965?gclid=Cj0KCQjwm6KUBhC3ARIsACIwxBhDaF2K9vTrhr4lTWNP-l3vxYgxWJZwK5n5Be9-TpcS9jbHR0vhFuEaAtoIEALw_wcB
  *   Then I tried adding a one wire temperature sensor.  I’m struggling with the code on this.
  *   If I read the one wire in the setup(), I get a good reading.
  *   When it’s the sensor loop, it fails
  *   I included my example code.

Questions/Answers
I do have wifi at a base station. (it is not super stable)
The tubing is sealed under vacuum, I found the MPRLS.  I connect small tubing from the SAP tubing to the MPRLS port.  This is working well.

Next steps.

  *   Get temp working
  *   Customize web pages for end user
I’m not very experienced with the docker process and I am struggling to find the web pages and code on the dms-lite pi server.

Again thanks for the assistance and interest.
Thanks

Jeff

hawk4031, 2022-05-21:
@ja2wiles I'll work up some documentation and send it your way on how to accomplish this. I will also make sure we have a detailed Development section in the DMS Lite repository documentation.

closed 2022-06-30 by tqrahman