ClusterDuck-Protocol/ClusterDuck-Protocol #305: Add support for CubeCell GPS-6502

BrentonPoke, 2023-05-22:
The [CubeCell GPS-6502]https://heltec.org/project/htcc-ab02s/) is an SX1262 board and has GPS capability, but We just need LoRa and OLED display working for now. Explicitely supporting GPS within the CDP may come at a later date, but we can just use this pinout to get the board's necessary functions for now.

[image]

## Notes

- SX1262 chips appear to [require a different sync word](https://forum.lora-developers.semtech.com/t/sx1272-and-sx1262-lora-sync-word-compatibility/988/2) than SX127X. The respective sync words are 

> SX127X publc,0x34 private,0x12
> SX126X publc,0x3444 private,0x1424

- Whoever implements this should have 2 of them to start testing, then see if they can communicate with SX1276 boards we already support. Testing a PR for this may cost money.

amirna2, 2024-02-29:
This is actually available in the CDP refactor branch, but not currently part of the upcoming 3.7 release.
We can target 3.8 release. It would be good to have another non ESP32 board supported.

amirna2, 2024-03-06:
CDP 4.0.0 added the pinout for this board, but it is not yet added to the build system.
I have one which I was able to use with my old version of CDP.

I would however hold off adding any new boards to CDP at this point, until we find a way for external boards to be added without having modify board support in CDP itself as this would lead to the unsustainable situation we had previously.

faradaym, 2024-08-26:
Resolved in cdp 4.0.0