ClusterDuck-Protocol/ClusterDuck-Protocol #307: Add a character counter to the Captive Portal text areas

amirna2, 2023-06-26:
**Describe the bug**
Not really a bug, but some UX improvement.
The captive portal text areas could have a character counter to indicate a limit, since we have just about 200 bytes limit for CDP package data section.

Here is a example of how to do this in the embedded HTML code using the `maxlength` attribute

```
<textarea class="textbox comments textbox-full" maxlength=200 name="message" id="commentsInput" cols="30" rows="2" onkeyup="updateCounter(this)">
</textarea>
                    Remaining characters: <span id="counter">200</span>

                    <script type="text/javascript">
                    function updateCounter(textarea) {
                        var currentLength = textarea.value.length;
                        var maxCharacters = 200;
                        var remainingCharacters = maxCharacters - currentLength;
                        document.getElementById("counter").innerHTML = remainingCharacters;  
                    }
                    </script>
```

tqrahman, 2023-08-15:
do you think max length of 200 be suffice for the 200 byte limit? or should we have a smaller limit like 140?

tqrahman, 2023-09-11:
added 200 character limit. It can be changed based on preferences.

closed 2023-09-11 by tqrahman