ClusterDuck-Protocol/ClusterDuck-Protocol #341: Umbrella issue: Bucket#1: Board Definitions / APIs

krook, 2024-02-12:
# Board definitions APIs

Description

## Part 1

- [ ] Create a separate repo/branch?
- [ ] Task 2
- [ ] Task 3

amirna2, 2024-02-19:
@krook I added more details for this task. It's a pretty straightforward once we start going down the list above

amirna2, 2024-02-22:
Note: this task will be on hold until we resolve current build issues found in CDP 3.6.1 when attempting to build non ESP32 boards.

amirna2, 2024-03-04:
I created a PR we can merge as part of the 4.0.0 integration branch.
Reduced the list of supported boards to:
```
heltec_wifi_lora_32_V2 (lora sx1276)
heltec_wifi_lora_32_V3 (lora sx1262)
ttgo_t_beam_sx1262 (lora sx1262)
heltec_cubecell_board v2 (lora sx1262)
heltec_cubecell_gps (lora sx1262)
rocket scream Mini Ultra Pro v3 (arduino zero SAMD21 core) (lora sx1276)
```

krook, 2024-03-06:
Looking good. Thanks for leading this @amirna2!

amirna2, 2024-03-06:
Closed this ticket as the tasks were completed for 4.0.0

closed 2024-03-06 by amirna2