ClusterDuck-Protocol/ClusterDuck-Protocol #355: Review onboarding documentation to ensure we match the new implementation

amirna2, 2024-03-04:


amirna2, 2024-03-04:
Adding a new board support in CDP requires defining their pinout as related to the LoRa Radio.
Each board has a header file that defines these pins and the header file gets included in the `cdpcfg.h` conditionally based on what the ARDUINO directive is set to.

BrentonPoke, 2025-02-16:
So I was informally assigned this task but I wanted input about _where_ it should be documented. I think the Wiki here on Github is sensible, but it's understandable that some will just go straight to code examples as the wiki feature is relatively new compared to the other Github features. It's years old, but i'm not sure how many people actually consult repository wikis.

closed 2025-04-07 by faradaym