ClusterDuck-Protocol/ClusterDuck-Protocol #368: Update each file with a copyright header

krook, 2024-03-08:
Based on the conversation in Discord, one of the things we should do is update all our files with a copyright header, such as this:

```c
// SPDX-FileCopyrightText: © 2018-2024 The ClusterDuck Protocol Authors
// SPDX-License-Identifier: Apache-2.0
```

For more context on how IP works in CDP:

* The Linux Foundation is the neutral home of all CDP code, trademarks, accounts, domains, etc. This ensures continuity of the CDP entity for anyone with an interest in it in case the current maintainers all hit the lottery and go live off the grid. See the [Project Charter](https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/blob/master/ClusterDuck-Protocol-Technical-Charter.pdf) for more details.
* The copyright of code is owned by the authors, "The ClusterDuck Protocol Authors" and/or anyone who contributes code they are allowed to provide (which they attest through the [Developer Certificate of Origin](https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/blob/master/DEVELOPER_CERTIFICATE_OF_ORIGIN.md) signoff on each commit). See sections 7a and 7d in the Charter.
* The copyright owners grant others the rights to use the code terms of use under the [Apache 2 License](https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/blob/master/LICENSE). For example, if the project is forked, copyright notices must be retained.

amirna2, 2024-03-08:
Does this apply to every file in the repository or only"code" source file?
I thought having the LICENSE.md file in the repo would be sufficient

krook, 2024-03-08:
It's a good question. Copyright can be implicit and the `LICENSE` file and a `NOTICE` in the home directory can be enough but the charter suggests it should be in each file (at least new ones contributed). 

I think the more practical part of it being in each file is that the SPDX part is machine-readable, for example if GitHub Copilot wants to know what it can do with a file it finds or if someone forks and copies just one file and puts it somewhere else.

amirna2, 2024-03-08:
Good catch! We need to update the PR template as well.

amirna2, 2024-03-08:
I created a short python script to perform batch update. It found 115 files need to have the SPDX headers.
The script is very basic and it's hardcoding the SPDX strings. Good enough for now.
We can make it fancier later with some CLI (maybe read the strings from a config file of something).

amirna2, 2024-03-08:
Created a new ticket for updating Doxygen header and footer. And also add the SPDX headers

krook, 2024-03-08:
For the documentation files we may want to use the Creative Commons license mentioned in the Charter