ClusterDuck-Protocol/ClusterDuck-Protocol #383: Remove Chat

tqrahman, 2024-03-11:
Remove the Chat system from all files.

amirna2, 2024-03-11:
@faradaym 

This is what I was referring to during the TH. Where chat calls into the `Duck::sendData()` method to actually send the CDP Packet containing the chat message, across the mesh.

This is really a terrible design and in order to make do this, some bad design decisions were made like now all methods in DuckRadio were made `private` and we have 5 CDP classes added a friends (including DuckNet)

```
  webServer.on("/chatSubmit.json", HTTP_POST, [&](AsyncWebServerRequest* request) {
    int err = DUCK_ERR_NONE;

    std::vector<byte> message;
    std::string clientId = "";

    AsyncWebParameter* p = request->getParam(0);
    std::string msg = p->value().c_str();
    message.insert(message.end(), msg.begin(), msg.end());
    logdbg_ln("chatSubmit: sending message. length: %d",message.size());
    std::vector<byte> muid;

    err = duck->sendData(topics::gchat, message, BROADCAST_DUID, &muid);
    addToChatBuffer(duck->buildCdpPacket(topics::gchat, message, BROADCAST_DUID, muid));

    switch (err) {
      case DUCK_ERR_NONE:
      {
        request->send(200, "text/html", "OK.");
      }
      break;
      case DUCKLORA_ERR_MSG_TOO_LARGE:
      request->send(413, "text/html", "Message payload too big!");
      break;
      case DUCKLORA_ERR_HANDLE_PACKET:
      request->send(400, "text/html", "BadRequest");
      break;
      default:
      request->send(500, "text/html", "Oops! Unknown error.");
      break;
    }
  });
  ```
```
class DuckRadio {
  friend class Duck;
  friend class DuckDetect;
  friend class DuckLink;
  friend class MamaDuck;
  friend class PapaDuck;

private:
  // Everything is private to force Duck (and Duck descendants) to be the only
  // way to interact with the radio. There should only be one Duck per sketch
  // so that arbitrary pieces of code cannot interfere with the radio. Also,
  // Duck does things like recording the outgoing MUIDs so that it can wait for
  // acknowledgments to those MUIDs.

  DuckRadio();
```  
  
  
So once chat is removed we need to take care in reverting all of these other changes in how they were in CDP 2.x
It is also critical that we add DuckRadio unit tests (which is only possible once we redefine the class to contain public methods as it was before).

faradaym, 2024-08-15:
Changes to DuckRadio were unrelated to chat, chat was removed in #390

closed 2024-08-15 by faradaym