ClusterDuck-Protocol/ClusterDuck-Protocol #388: Remove some pages from the Wiki

amirna2, 2024-03-13:
**IMPORTANT**  
Check the Wiki, developer's guide or ask the community on Slack before submitting a new ticket. You might find a solution to your issue there.

Also, please make use of [markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) to format your issue. This will make it easier for us to understand and help you. 

[image]

The Developer resources pages could be removed.
- Design Document
- Message Handling
- Versioning
- Travis - Build Server

Versioning doesn't need explaining. We could simply state we are following semver and add a link to https://semver.org

Additionally we should link the Developer's Guide in that Developer Resource section.

TimoWielink, 2024-03-18:
I will also take a look at this, start with outlining if we want to archive documentation for V3 and separate based on versions

tqrahman, 2024-04-02:
The table of contents is simplified!

closed 2024-04-02 by tqrahman