ClusterDuck-Protocol/ClusterDuck-Protocol #399: Review and refactor DuckCrypto

amirna2, 2024-03-16:
As currently implemented DuckCrypto is not really usable. Keys are hardcoded and API is incomplete and does not support end-to-end encryption.

Much effort remains in designing encryption in a way that fits with the network protocol and specifically how devices can be included/excluded from a cluster.

TimoWielink, 2024-03-18:
Do we want to remove this feature, or work on a new implementation

TimoWielink, 2024-03-18:
**From meeting**
- Encryption is always enabled by default
- Need to re-implement the way we create and assign keys
- How do you set encryption for your cluster 

- [ ] Make Crypto into a class
- [ ] Create a system for assigning and creating clusters (with same encryption)

faradaym, 2025-09-25:
DuckCrypto is being removed in CDP v5 and will be completely redesigned

closed 2025-09-25 by faradaym